Bodmer/TFT_eSPI #829: Getting a 3.5 Rpi Display Working - Works on RaspberryPi with LCD35-Show Driver

Tysonpower, 2020-11-18:
[image]
I currently trying to get a Raspberrypi 3.5 Display working on a ESP32 V1 DevKit from DOIT but i only get a white Screen so far.
I know that there where many discussions about these display, so i hope i have one that will work in the End.

I've read a lot of Forum Posts and connected everything like it is stated in the Image under Tools as well as in the 
[Arduino Forum](https://forum.arduino.cc/index.php?topic=499672.0).
I also tested the Display with the Waveshare 3.5 Driver on a RaspberryPi and it worked on the first try.
With that it should have the ILI9486 Driver IC like the Waveshare Display right? Otherwise it shouldn't work with the LCD35-show Setup File from Waveshare, at least in my theory.

[image]
The DIagnostics also look good, so the Config is loaded:

```
TFT_eSPI ver = 2.3.4
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9486
Display width  = 320
Display height = 480

MOSI    = GPIO 23
MISO    = GPIO 19
SCK     = GPIO 18
TFT_CS   = GPIO 15
TFT_DC   = GPIO 2
TFT_RST  = GPIO 4
TOUCH_CS = GPIO 22

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 20.00
Touch SPI frequency   = 2.50
```

I hope someone has an Idea,
regards Manuel

Bodmer, 2020-11-18:
Have you set RPI_DISPLAY_TYPE in the setup file, [example line here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L36). This is needed as the RPi displays use 16 bit commands.

Tysonpower, 2020-11-18:
> Have you set RPI_DISPLAY_TYPE in the setup file, [example line here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L36). This is needed as the RPi displays use 16 bit commands.

Hi bodmer, i use your Setup_11 File with the following Content in it. I also checked that this file is used by changing the SPI rate back and forth.

```
// See SetupX_Template.h for all options available
#define RPI_DISPLAY_TYPE
#define ILI9486_DRIVER // 20MHz maximum SPI

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)
//#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST

#define TOUCH_CS 22     // Chip select pin (T_CS) of touch screen

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

#define SPI_FREQUENCY  20000000

#define SPI_TOUCH_FREQUENCY  2500000
```

Bodmer, 2020-11-18:
Run the "Test_Touch_Controller" sketch to verify the SPI bus is connected correctly. The output should look like this, with high Z values when touched:
```
x: 0     y: 3861     z: 13 
x: 0     y: 3873     z: 12 
x: 1518     y: 2225     z: 1011 
x: 1664     y: 2169     z: 1753 
x: 1681     y: 2154     z: 1933 
x: 1679     y: 2142     z: 1955 
x: 1696     y: 2153     z: 1900 
x: 0     y: 3848     z: 14 
x: 0     y: 3889     z: 15 
x: 0     y: 3903     z: 14 
x: 0     y: 3894     z: 14 
```
I have connected up an identical looking display using setup 11 and it is working fine, so I am not sure what to suggest...
[image]

Bodmer, 2020-11-18:
I see you have connected power to the 3.3V input. If you have conneted to the ESP32 VIN pin then be aware that is 5V and may damage the display. My display does not work from 3V3, even when connected to the pin you have used, I have to connect 5V (not 3.3V) for the display to work, but that is to the other corner 5V pin that you have not used as in this diagram.
[image]

Yes, I have connected exactly as in the diagram below. Note that I have used different 0V pins to you as in the diagram, (but that is not the problem, you have used a different GND pin.

Tysonpower, 2020-11-19:
IT Finally WORKS! @Bodmer 

I don't know why those guys in China did that, but my Display needs to be connected to 3.3V AND 5V to work.
If you only connect 3.3V (Pin 1) as i did before, the TFT lights up BUT only when 5V (Pin 2) is connected as well the Display works.

I tested a lot around with a RaspberryPI 3B to find that out, but hey now it works fine including the Touchscreen.

Anyway, thanks for your Help Bodmer, maybe you can add a short notice to the Readme that some Displays seem to be that strange, so other People don't have to struggle as i did :)

[image]

closed 2020-11-19 by Tysonpower

Bodmer, 2020-11-19:
I expect you will find you can now disconnect the 3,3V power feed to the RPi display because it has an onboard 3.3V regulator powered off the 5V line.

Tysonpower, 2020-11-19:
> I expect you will find you can now disconnect the 3,3V power feed to the RPi display because it has an onboard 3.3V regulator powered off the 5V line.

Nope. as i said, both need to be connected. When i disconnect the 3.3V line the TFT goes black. It doesen't make sense to me either, but this Display needs both lines despite the Regulator on the Back.