Bodmer/TFT_eSPI #830: HIgh speed SPI support for MHS-4.0inch Display-B (RPi 480x320 ST7796S)

1ecu, 2020-11-19:
Hi,

I’d like to present yet another RPi TFT puzzle: ESP32 (TTGO T8) with MHS 4.0” RPi 480x320 ST7796S high speed SPI (125 MHz)  http://www.lcdwiki.com/MHS-4.0inch_Display-B . 

Pins in my setup are TFT_MISO 19, TFT_MOSI 23, TFT_SCLK 18, TFT_CS   5, TFT_DC   27, TFT_RST  -1, TOUCH_CS 22. In order to get the display working I need to set

#define RPI_DISPLAY_TYPE
#define ST7796_DRIVER
#define SPI_FREQUENCY  35000000

MHS works normally up till about 35 Mhz. At 36 - 39 Mhz stray pixels or other disturbances can show up. White screen at 40 MHz . Touch is ok.

I’m not sure if this particular display model has been discussed earlier here or on some other forum but I nevertheless recall also someone else mentioning that even if TFT_eSPI supports MHS RPi shields they might still be able to work only at normal or even lower SPI speeds (albeit examples run ok). Well, I’m aware of the "normal" / recommended SPI speeds but I think in principle it should be possible to reach around 80 MHz with ESP32 – especially with this kind of high speed TFT.

So assuming my MHS has the advertised capabilities I’m wondering if the speed issue has something to do with the fact that when RPI_DISPLAY_TYPE is defined then at least in processor files (TFT_eSPI_ESP32.h in this case) some of the data transfer actions seem to get throttled (there are comments like “// RPi displays need a slower DC change” etc.)? As far as I understand this happens regardless of which driver or SPI speed is defined for a RPi display.

Bodmer, 2020-11-19:
This is almost certain to be a signal integrity problem.

To work at high SPI frequencies the signal wires must be kept as short as possible, no more than 100mm. I have a breadboard setup identical to yours and if I icrease the wire length, to 150mm, particularly on the SCK signal, then occasional streaks occur at 80MHz. This is down to edge ringing on the signal. I suspect if you hold the SCK wire between fingers the streaks get worse, this is an indicator that the clock signal is suffering from distortion.

Use at least 2 GND wires from the ESP32 to the display, this helps reduce ground bounce.  Ideally in high speed designs a ground plane between the display and processors would be used. Make sure the power feed to the ESP32 and display are adequate, higher rates demand higher power peaks.

1ecu, 2020-11-19:
Well, it was indeed signal related so thanks for good advices :-) Interestingly, straightening and shortening the wiring from 300 to some  100 mm didn't basically change anything but in this case the trick was to connect all three T8 ground pins to tft. This alone allow 70-75 MHz SPI speed even with longer wires.

Bodmer, 2020-11-20:
Valid high SPI clock rates for the ESP32 are 26.6667MHz (use 27MHz), 40MHz and 80MHz.  Other frequencies are rounded down to those values, e.g. 79MHz is rounded down to 40MHz. This can be shown to be the case using the graphicstest example.

Glad to hear it is working!

closed 2020-11-20 by Bodmer