Bodmer/TFT_eSPI #831: Using viewport within a sprite. Howto?

ha-ina, 2020-11-20:
I am using a full-screen sprite  "spr" (320x240 pixels) to draw some graphics into the sprite before I push it onto the tft to avoid flickering on the screen.
This works fine until  I try to set a viewport within the sprite for clipping 
 spr.setViewport(x,y,w,h,false)
 //draw some lines ... into the sprite via sp.drawLine(...)
spr.resetViewport();  
// and finally push the sprite to the tft screen

// the lies appear on the screen but the clipping to the viewport are  is not done

What am I doing wrong, or is it not possible to set a viewport within a sprite?

I appreciate any help or comment.
Thank you

Bodmer, 2020-11-20:
The viewport capability has only recently been added and at the moment the viewport only apply to the TFT screen.

I will be expanding this capability to the Sprites, this should be relatively easy, but I am proceeding cautiously with incremental updates in capability.

ha-ina, 2020-11-21:
Thanks for your quick response.
 It is good to hear you are planning to implement that feature as an enhancement.
Another useful feature would be the possibility to push a rotated sprite not only to the screen, but also onto another sprite.
This would be of great help for my intended application.

Bodmer, 2020-12-01:
The update has been made. Viewports are now available for Sprites.

Please test on your setup.

ha-ina, 2020-12-02:
Thanks a lot for your prompt action.
I will try it as soon as I have downloaded and installed your latest version.
This might take some time since I had done some local additions (e.g. drawArc() )  which I want to preserve before installing your latest version.
I will let you know about the success when I have tried your changes.
You mentioed in another posting that you have compared the speeds for pushToSprite() for different combinations of source and destination bpp, including 8-bpp to 8-bpp. I had tried that, but then realized from the source that this is not implemented, and I used 16-bpp to 8-bpp instead.
Is it true that 8-bpp to 8-bpp is now possible? (Would be nice to have).

Bodmer, 2020-12-02:
Yes, 8bpp to 8bpp is now supported.

Bodmer, 2020-12-02:
It is easier if you put your own functions in an extension library https://github.com/Bodmer/TFT_eFX

closed 2020-12-04 by Bodmer