Bodmer/TFT_eSPI #832: ili9486 working schematic

clear-sky, 2020-11-21:
hi all and thanks for this great driver
i have 3.5 inch ili9486 
http://aitendo3.sakura.ne.jp/aitendo_data/product_img/lcd/tft2/Z350IT002/Z350IT002_aitendo.pdf
and want to make a pcb for use it in spi mode and found these schematics
[image]
[image]
[image]
my question is which is compatible with this driver

pipi61, 2020-11-21:
https://forum.arduino.cc/index.php?topic=443787.90

Bodmer, 2020-11-22:
I do not recommend any of those circuits because they are poorly designed and rely on a logic race hazzard. This means they only operate over a limtied frequency range.

I suggest you connect your processor direct to the display, either with a SPI interface or 8 bit parallel interface. Study the data sheet and build a wired prototype before commiting to a PCB.

closed 2020-11-22 by Bodmer