Bodmer/TFT_eSPI #833: Try both ILI9488 and ILI9481 3.5" LCD. ILI9481 doesn't work.

garryhca, 2020-11-23:
MCU: ESP32. 

User_Setup.h, I chose one of the following driver accordingly.  Use the same SPI bus for both screens.
#define ILI9481_DRIVER 
#define ILI9488_DRIVER 

Program: meters example.

ILI9488 works fine. ILI9481 doesn't work, nothing shows up. Please fix the ILI9481 driver.

Thank you,
Garry

Bodmer, 2020-11-23:
SPI or parallel interface? SPI diver is working fine with my display.
Provide a link to the vendors web page that shows the display.
Post pictures of your setup and describe the connections you have used.
Some vendors send ILI9486 screens instead of ILI9481, so try that driver.

garryhca, 2020-11-23:
Both screens have the same FPC 40P pins definition. The only difference is the driver chip inside. 
I attach the vendor supplied schematic for the screens.  IM0, IM1, IM2 are pulled high to 3.3V to support 4Wire SPI. 
https://1drv.ms/u/s!AqPQ3Z1I41ct90gu44yba1VtWTo0?e=F6kRhd

Or do you use 3Wire SPI mode?

#define ILI9481_DRIVER (9481 LCD doesn't work)   or  #define ILI9488_DRIVER  (9488 LCD works)
#The following settings are the same for both LCD screens.
#define TFT_MISO 12
#define TFT_MOSI 13
#define TFT_SCLK 14
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   27  // Reset pin (could connect to RST pin)

I will try ILI9486 driver later today.

Bodmer, 2020-11-23:
Check if the sketch is uploading, the use of pin 2 can prevent this. 4 wire SPI is correct.
Make sure the backlight LED Cathode pins are wired to GND, the schematic has an error here.
Check the ribbon is the right way up, metal contanct down on PCB side.
Beware: some ribbon cables on some displays is the opposite sequence to the breakout board (ie ribbon pin 1 goes to FPC PCB adapter pin 40, and ribbon pin 1 goes to FPC adapter pin 1 !!!

Do not use wires longer that 100mm, use multiple ground connections from display to ESP32. Keep SPI clock rate low (10MHz) for testing initially.
Post a picture of your setup.

garryhca, 2020-11-23:
The meters sample Sketch uploads fine every time. 
I built my own PCB. The distance is very short between PFC and ESP32. Less than 50mm. 
Backlight LED is controlled by a GPIO. I turn it on in the setup() section. Backlight LED Works fine. 
As I said in the previous posts, both 3.5" screens are from same manufacturers. Same FPC pins definitions.
The only difference is ILI9481 one is IPS. ILI9488 one is TN. Hopefully we can make the ILI9481 one works since IPS is a lot better.
Here is a photo shows the ILI9481 is working. 
https://1drv.ms/u/s!AqPQ3Z1I41ct90llPnDSu6hr-FdS?e=t6V8wt

I will test the IPS ILI9481 again tonight when I am home. 
1. Try ILI9486 driver
2. Change SPI Speed in TFT_eSPI.h, correct?
/***************************************************************************************
**                         Section 3: Interface setup
***************************************************************************************/
#ifndef TAB_COLOUR
  #define TAB_COLOUR 0
#endif

// If the SPI frequency is not defined, set a default
#ifndef SPI_FREQUENCY
  #define SPI_FREQUENCY  20000000  //(changed to 10000000 ?)
#endif

garryhca, 2020-11-24:
Hi Bodmer,
I just tested with #define SPI_FREQUENCY 10000000, ILI9486 driver and ILI9481 driver.  None of them work.
I just asked the vendor to send me another one just in case of a defective screen. Will test again.
The ILI9488 one works perfectly on the same board. Double check again on their datasheet. All pins definitions are exactly the same on both ILI9481 and ILI9488 LCD screens. No any difference. 

Here is the sample code for ILI9481 SPI mode they sent me. I think it is coded with Keil. I don't know how to read it. Maybe you can take a look at their driver part.
https://1drv.ms/u/s!AqPQ3Z1I41ct-AEX1w3v9CkMVY10?e=aNs8Tk 

Thank you,
Garry

Thanks,
Garry

garryhca, 2020-11-25:
Hi Bodmer,
I just received and tested with another ILI9481 and ILI9486 3.5" screens the vendor sent me.
1. Tried ILI9481, ILI9486 and ILI9488 drivers on ILI9481 screen. Still no display.  
2. Tried ILI9486 and ILI9488 drivers on ILI9486 Screen. ILI9486 driver shows nothing. ILI9488 driver can display, but the color is green only. Like this: https://1drv.ms/u/s!AqPQ3Z1I41ct-APsXbWHu1wIKaT0?e=vNMzzO
3. ILI9488 screen still works great with ILI9488 driver.
https://1drv.ms/u/s!AqPQ3Z1I41ct-AL3XFQtzJO5h0FT?e=puhecz

Now, I guess there are two possibilities.
1. I need to use different IO for connection since you said your ILI9481 works fine with ESP32?  What IOs do you use for the connection?
2. Maybe, it is ILI9481 and ILI9486 drivers issue with ESP32?

Thanks,
Garry

Bodmer, 2020-11-25:
I have updated the library for the ILI9481, it appears this display driver required 18 bit colours when used in SPI mode. Note also that the data sheet quotes 10MHz maximum for the SPI clock. I found it runs OK at 15MHz, but not 16MHz or higher.

This change has been tested with ESP8266 so far but I have applied the same changes to the ESP32 code so please try it.

garryhca, 2020-11-26:
Thanks for your quick update. ILI9481 works great now with ESP32! Just tested ILI9486 again, but it still doesn't work. 
ILI9488 OK.
ILI9481 OK.
ILI9486 NO.

Bodmer, 2020-11-26:
I have added some [different init options](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ILI9481_Init.h#L8-L13) for the ILI9481. A different option may be needed on your display to get the colour balance right.  Run the TFT_Rainbow480 sketch and choose an option that gives roughly equaly red, green, blue, yellow, magenta and cycan bands. You probably wont get it perfect as screens vary. I found option 5 gave the best colour balance.

No changes have been made for ILI9486 yet, so no point in trying it. I do not have a SPI ILI9486 but will make the same changes as for ILI9481 in the hope that it works.

Bodmer, 2020-11-26:
Thanks for reporting the ILI9481 is working for you.

I have updated the ILI9486 code with a best guess at the changes needed, open another issue if it does not work.

closed 2020-11-26 by Bodmer