Bodmer/TFT_eSPI #834: just want to say what a great library

ldijkman, 2020-11-24:
not an issue but i do not now how to make a compliment comment
just want to say what a great library
saw a demo at https://youtu.be/rq5yPJbX_uk
the touch calibration looked perfect in that video
playing with esp32 wroom 32 devkit and 4" st7796 spi display
want to use it on my Thermostat https://youtu.be/lny7J7KXk9w
Thank you for this superb library
greet luberth

Bodmer, 2020-11-24:
Thank you for the positive feedback and link to your project.  The ST7796 is my favorite display, it was a good choice for your project.

ldijkman, 2020-11-24:
Spend some hours to get the touch working on esp wroom 32 dev kit 38 pin  on tft touch spi st7796
calibration only works ok  in screen orientation 0 then all 4 corners react ok to touch 
and demos work correct with touch  in orientation 0  (button on/of and keypad)
in other orientation => 1 corner is not reacting on touch 
and or it looks like i have to touch another corner as where the arrow is

ldijkman, 2020-11-25:
looks like i have to change this 
otherwise the calibration process is not working mostly on one corner

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2000000  // changed was   2500000

ldijkman, 2020-11-27:
try to design a PCB 
https://oshwlab.com/l.dijkman/esp32-dev-kit-38-pin-to-spi-touch-tft

closed 2020-12-05 by Bodmer