Bodmer/TFT_eSPI #835: Example for serialplotter

yilmaz1964, 2020-11-25:
Thank All for the  effords, 
This the only Library that worked for me.
May I ask if it is possible to make SerialPlotter example code .
It would help me with my ecg project.
Thanks U.

Bodmer, 2020-11-25:
No special code in the library is required to use the Arduino IDE's Serial Plotter. You just need to send the values to the serial port and select the Arduino IDE plotter window.

yilmaz1964, 2020-11-25:
Hi
Thanks for the fast reply.
Sorry for my bad English.
In the example codes there is an example TFT_eSPI/320x240/TFT_Terminal.ino
which works perfectly.
Is it possible to write code for the tft to emulate a Terminal plotter?
like the serial plotter inside the Arduino ide, but for the tft screen.
Thanks Greetings from Belgium.
Yilmaz,

Bodmer, 2020-11-25:
Use the Sprtie_scroll_4bit example as a starting point. This draws a scrolling graph.

You will need to parse the ASCII stream into integer values then plot them. Typically the stream would be comma separated numbers.

The library examples are written to show features of the library and need to be expanded and modified for projects.

closed 2020-11-25 by Bodmer