Bodmer/TFT_eSPI #836: How to port this library to ILI9225

pierbout, 2020-11-25:
Hello,

I tried to adapt the library for ILI9225 (2 or 2.2inches TFT SPI screen 176x220pixels), which is very close from ILI9341 but with different resolution. I updated for example GUIslice with success, using this metod, but here it failed (the screen remains white).

Here are the files created or updated.
[TFT_eSPI-ILI9225.zip](https://github.com/Bodmer/TFT_eSPI/files/5599415/TFT_eSPI-ILI9225.zip)

Do you have any suggestion? I don't know for exemple which values to use for the frequencies

eSPI-read-user_setup displays indicates a wrong value of display driver (I selected ox9225, but I don't know if this is the correct value, as I didn't fid what it corresponds to):

[code]
Transactions = Yes 
Interface    = SPI 
Display driver = -28123 
Display width  = 176 
Display height = 220 

MOSI    = D-1 (GPIO 51)
SCK     = D-1 (GPIO 52)
TFT_CS   = D-1 (GPIO 28)
TFT_DC   = D-1 (GPIO 24)
TFT_RST  = D-1 (GPIO 26)

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency =  ? MHz 
[/code]

Bodmer, 2020-11-25:
I have added ILI9225 support to the GitHub copy.

Please test it, I do not have an ILI9225 display to try it.

pierbout, 2020-11-26:
Thank you for being so quick. 

From your master:
-In User_Setup.h: I added #define ILI2995_DRIVER
-In User_Setup.h: I adapted the pins numbers to my board
-In User_Setup_Select.h: I adapted PIN_MOSI and PIN_SCLK  to my board (I suggest to group them in the same macro, or at least the same file if possible)

--> I get a compilation error:
/home/pierre/Projets/Arduino/libraries/TFT_eSPI-master/TFT_eSPI.cpp: In member function 'void TFT_eSPI::setWindow(int32_t, int32_t, int32_t, int32_t)':
/home/pierre/Projets/Arduino/libraries/TFT_eSPI-master/TFT_eSPI.cpp:2869:21: error: 'TFT_CASET' was not declared in this scope
   DC_C; tft_Write_8(TFT_CASET);

Notes:
-"#define TFT_CASET" in indeed missing in TFT_Drivers/ILI9225_Defines.h  (I don't know what it corresponds to)
-There is no User_Setups/*ILI9225* file. Is it normal?

Thanks for your help

pierbout, 2020-11-26:
If I copy lines 34 to 62 of TFT_Drivers/ILI9341_Defines.h  in TFT_Drivers/ILI9225_Defines.h , I get on the screen a fog of coloured pixels (which is an evolution from the white screen!)

Bodmer, 2020-11-26:
The update involved some guesswork so I am not surprised it is not working correctly.

PIN_MOSI and PIN_SCLK are ESP8266 only and do not need to be defined in the setup file, [only define the 3 pins in the default setup](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L156-L158) for that processor as the other SPI pins are fixed.

Bodmer, 2020-11-26:
I have had a rummage and found a small ILI9225 screen, it will be a few days before I can play with it and see if I can get it working correctly.

pierbout, 2020-11-26:
Thank you Bodmer.
I tried to comment PIN_MOSI and PIN_SCLK in User_Setup_Select.h, then TFT_BL, TFT_SCLK and TFT_MISO in User_Setup.h, I get the same random color pixels on the screen. I guess I'm not far, as I ger color, but there are probably one or two parameters missing somewhere. Don't hesitate if you need me to test something.

Bodmer, 2020-12-01:
I have added ILI9225 support, this has been tested on  a SPI 176 x 220 pixel display.

Please test on your setup.

pierbout, 2020-12-01:
Good job! It works great!

Thanks a lot.

closed 2020-12-02 by Bodmer