Bodmer/TFT_eSPI #837: RM68090 compatibility

michaelwilkinson1992, 2020-11-25:
### Feature Request or Enhancement
*Please describe what you would like to see GUIslice do*: ??? Any chance of adding drivers for the RM68090 board?
https://www.banggood.com/Geekcreit-UNO-R3-Improved-Version-+-2_8TFT-LCD-Touch-Screen-+-2_4TFT-Touch-Screen-Display-Module-Kit-Geekcreit-for-Arduino-products-that-work-with-official-Arduino-boards-p-1428291.html?utm_source=googleshopping&utm_medium=cpc_organic&gmcCountry=DE&utm_content=minha&utm_campaign=minha-de-en-pc&currency=EUR&createTmp=1&utm_source=googleshopping&utm_medium=cpc_bgs&utm_content=frank&utm_campaign=frank-ssc-de-all-en-0608feed&ad_id=440288844860&cur_warehouse=CN

### Additional context
*Currently works using MSCFriends library, I suspect it might be possible to use the ili9341 driver with little effort if the display ID is forced??*

Bodmer, 2020-11-25:
Use the ILI9341_DRIVER option. That should work fine.

michaelwilkinson1992, 2020-11-26:
Works for the 2.4“ (colors and x/y axis are inverted though) but on the 2.8“ I just get a white screen.

I think* I need to somehow force the display ID, as when I use the MCUfriends library, it works on the 2.8“, but the 2.4“ only works when the display ID is forced.

What are your thoughts, any way to try it?

michaelwilkinson1992, 2020-11-26:
I would also like to add, I am trying to use your repo in conjunction with GUISlice, but I’m having some issues there. I’m not 100% sure where to post, I’ve opened an issue on their repo regarding this particular display, it may or may not help the situation here, but here is a link anyway. https://github.com/ImpulseAdventure/GUIslice/issues/297

Bodmer, 2020-11-26:
Many cheap displays are available that work with TFT_eSPI, so I suggest you buy one of them as I do not have the time to try and get all dispays working with the library. Especially when I do not own a display of that type.

I am unable to resolve any issues with GUISlice so your best option is to raise an issue on that repository.

Sorry I can't be more helpful this time.

closed 2020-11-26 by Bodmer

michaelwilkinson1992, 2020-11-26:
Can you tell me where display ID‘s are defined, or if they are defined at all?

Bodmer, 2020-11-28:
The display driver type is specified by the user in the User_Setup.h file of the library. The driver specific initialisation files are in the TFT_Drivers folder of the library.