Bodmer/TFT_eSPI #839: Push sprite to another sprite

Paulie92, 2020-11-27:
Hello Bodmer,

Does TFT_eSPI library have any function, which can push one sprite into another sprite? I have a background sprite and I want to push into this sprite another one and after that the complex image send to the sreen with standard spr.pushSprite. I found out that the only way hove to done this is to use function pushRotated(&spr, angle), where you can point the other sprite, where this sprite will be pushed in (you have to use also spr.setPivot(x,y) to determin the location where the sprite should be pushed.

Is there any more elegant funcion, which can done this? Because I assume, that the function "pushRotated" does some rotation calculations which are not needed if I do not want to rotate the sprite.

The second question is how fast is one sprite pushed into another one. I thought that it will be very fast, bacause it is done only in she RAM, but it seems it is same fast as if the sprite is pushed directly to the TFT.

Thank you for any answers in advance.

Paulie92, 2020-11-27:
Oh, I found it: pushToSprite. I had old version of the library, sorry.

But can you please let me know, why the pushing of one sprite to another one is as "slow" as the pushing sprite to the TFT? I assumed that for examle pushing a sprite with 50x50pixels filled red to another sprite should be as fast as drawing a red Rect with the same size.

Thank you for the answer.

Bodmer, 2020-11-28:
The sprite copy should not be that slow.
Post a complete example showing the problem.

ha-ina, 2020-11-30:
I had some trouble pushing a sprite onto another sprite until I found out that pushing 8-bpp onto another 8-bpp is not implemented. Then using 16-bpp onto 8-bpp worked fine.
Is there a reason not to allow 8-bpp as the source for a 8-bpp destination?
( By the way: Thanks for your marvelous library!)

Paulie92, 2020-11-30:
You are right, only the pushRotated function is slower, pushToSprite works fast. My mistake, I am sorry. But this function unfortunately does not support the transparent color. Is there any problem with that, or you will just implement it later?

Bodmer, 2020-12-01:
Handling a transparent copy is planned but has not been added yet.

I ran some fill and copy time tests to see how this performed. 1bpp Sprites are slow due to bit manipulation and the coordinate rotation available for 1bpp Sprites. These 1bpp Sprites were intended to support ePaper displays which have a very slow refresh anyway.  The 16bpp to 8bpp Sprites have the overhead of colour translation hence will be much slower than driect byte copying.

```
// 50 x 50 pixel sprite to sprite copy times

// ESP8266
//  Source Destination  Fill dest      pushToSprite
//   16bpp    16bpp      41.1 us       39.8 us
//   16bpp     8bpp      23.1 us      289.6 us
//    8bpp     8bpp      23.2 us       31.0 us
//    4bpp     4bpp      19.1 us       27.8 us
//    1bpp     1bpp    1470.0 us     1700.0 us

// ESP32
//  Source Destination  Fill dest      pushToSprite
//   16bpp    16bpp      25.8 us       23.8 us
//   16bpp     8bpp      20.8 us      260.2 us
//    8bpp     8bpp      15.8 us       20.8 us
//    4bpp     4bpp      14.2 us       19.8 us
//    1bpp     1bpp    1336.0 us     1476.0 us
```

closed 2020-12-01 by Bodmer

Bodmer, 2020-12-02:
Transparent copy of a sprite (spr) to a destination sprite (dspr) has been added, function example is:
```
spr.pushToSprite(&dspr, x, y, transp_color);
```

Paulie92, 2020-12-07:
Great! Thank you for your effort. Really appreciate it. 

By the way, I am using in my sketch pushing sprite to TFT to create fast moving texts. As I recently downloaded a new version of your library, the text is moving about half a speed (so I assume it takes longer to push the sprite). The setting (SPI clock) is the same. The previous version of library was approx. one year old. Do you know where the problem could be? Did you made any changes which can cause this?

PS: best library ever!

Bodmer, 2020-12-07:
I see no obvious reason for the performance change. Post a complete sketch that demonstrates the scrolling and shows which functions you are using.

Paulie92, 2021-01-18:
I forgot to change the SPI frequency (I am using 40mHz)  when downloading the new version of the library, so it was only my mistake. Sorry for that.

Bodmer, 2021-01-18:
OK, no worries.