Bodmer/TFT_eSPI #841: Boing Ball Test Compile Error - Arduino IDE

Guerillah, 2020-11-27:
I've tried other examples and they all compile and upload correctly. The boing ball test has throws the same error every time I've tried.

[image]

Guerillah, 2020-11-27:
Error text:

Linking everything together...
"C:\\Users\\Davis\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-gcc" -nostdlib "-LC:\\Users\\Davis\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/lib" "-LC:\\Users\\Davis\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/ld" -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority -u __cxa_guard_dummy -u __cxx_fatal_exception -Wl,--start-group "C:\\Users\\Davis\\AppData\\Local\\Temp\\arduino_build_814478\\sketch\\boing_ball.ino.cpp.o" "C:\\Users\\Davis\\AppData\\Local\\Temp\\arduino_build_814478\\libraries\\TFT_eSPI\\TFT_eSPI.cpp.o" "C:\\Users\\Davis\\AppData\\Local\\Temp\\arduino_build_814478\\libraries\\SPI\\SPI.cpp.o" "C:\\Users\\Davis\\AppData\\Local\\Temp\\arduino_build_814478\\libraries\\FS\\FS.cpp.o" "C:\\Users\\Davis\\AppData\\Local\\Temp\\arduino_build_814478\\libraries\\FS\\vfs_api.cpp.o" "C:\\Users\\Davis\\AppData\\Local\\Temp\\arduino_build_814478\\libraries\\SPIFFS\\SPIFFS.cpp.o" "C:\\Users\\Davis\\AppData\\Local\\Temp\\arduino_cache_73596\\core\\core_612199147e23bfc61383090334c04571.a" -lgcc -lesp32 -lphy -lesp_http_client -lmbedtls -lrtc -lesp_http_server -lbtdm_app -lspiffs -lbootloader_support -lmdns -lnvs_flash -lfatfs -lpp -lnet80211 -ljsmn -lface_detection -llibsodium -lvfs -ldl_lib -llog -lfreertos -lcxx -lsmartconfig_ack -lxtensa-debug-module -lheap -ltcpip_adapter -lmqtt -lulp -lfd -lfb_gfx -lnghttp -lprotocomm -lsmartconfig -lm -lethernet -limage_util -lc_nano -lsoc -ltcp_transport -lc -lmicro-ecc -lface_recognition -ljson -lwpa_supplicant -lmesh -lesp_https_ota -lwpa2 -lexpat -llwip -lwear_levelling -lapp_update -ldriver -lbt -lespnow -lcoap -lasio -lnewlib -lconsole -lapp_trace -lesp32-camera -lhal -lprotobuf-c -lsdmmc -lcore -lpthread -lcoexist -lfreemodbus -lspi_flash -lesp-tls -lwpa -lwifi_provisioning -lwps -lesp_adc_cal -lesp_event -lopenssl -lesp_ringbuf -lfr -lstdc++ -Wl,--end-group -Wl,-EL -o "C:\\Users\\Davis\\AppData\\Local\\Temp\\arduino_build_814478/boing_ball.ino.elf"
C:\Users\Davis\AppData\Local\Temp\arduino_build_814478\sketch\boing_ball.ino.cpp.o:(.literal._Z5setupv+0x24): undefined reference to `TFT_eSPI::initDMA()'
C:\Users\Davis\AppData\Local\Temp\arduino_build_814478\sketch\boing_ball.ino.cpp.o:(.literal._Z4loopv+0x8c): undefined reference to `TFT_eSPI::pushPixelsDMA(unsigned short*, unsigned int)'
C:\Users\Davis\AppData\Local\Temp\arduino_build_814478\sketch\boing_ball.ino.cpp.o: In function `setup()':
C:\Users\Davis\AppData\Local\Temp\arduino_modified_sketch_78641/boing_ball.ino:68: undefined reference to `TFT_eSPI::initDMA()'
C:\Users\Davis\AppData\Local\Temp\arduino_build_814478\sketch\boing_ball.ino.cpp.o: In function `loop()':
C:\Users\Davis\AppData\Local\Temp\arduino_modified_sketch_78641/boing_ball.ino:166: undefined reference to `TFT_eSPI::pushPixelsDMA(unsigned short*, unsigned int)'
collect2.exe: error: ld returned 1 exit status
Using library TFT_eSPI at version 2.3.4 in folder: C:\Users\Davis\Documents\Arduino\libraries\TFT_eSPI 
Using library SPI at version 1.0 in folder: C:\Users\Davis\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI 
Using library FS at version 1.0 in folder: C:\Users\Davis\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS 
Using library SPIFFS at version 1.0 in folder: C:\Users\Davis\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPIFFS 
exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.

Bodmer, 2020-11-28:
I have compiled the example with no problems on Arduino IDE version 1.8.13 with ESP32 board package 1.0.4 and for the DOIT ESP32 DEVKIT V1 board. 

I suggest you load a clean install of the TFT_eSPI library 2.3.4 via the IDE library manager and try again.

closed 2020-11-28 by Bodmer

Guerillah, 2020-12-05:
I think this may have to do with me using an MHS 4.0 ST7796S display? The documentation in your readme states that DMA doesn't work with RPi type displays. Is there a reason for this? Will that be changed at some point? Thanks sir.

Bodmer, 2020-12-05:
The latest version on Github does allow DMA with that display.

Guerillah, 2020-12-05:
I'm not finding this to be true. I've tried on two different computers with fresh installations of Arduino IDE (v 1.8.13)  and TFT_ESPI(v 2.3.4). The display compiles and run your examples fine for everything but the DMA test where it will throw the errors in the original post. I have attached a .zip file with my user setup files along with the dump below.

[code]
TFT_eSPI ver = 2.3.4
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 7796
Display width  = 320
Display height = 480

MOSI    = GPIO 23
MISO    = GPIO 19
SCK     = GPIO 18
TFT_CS   = GPIO 15
TFT_DC   = GPIO 2
TFT_RST  = GPIO 4
TOUCH_CS = GPIO 22

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 80.00
Touch SPI frequency   = 2.50
[/code]

[TFT_eSPI_User_Setup.zip](https://github.com/Bodmer/TFT_eSPI/files/5647940/TFT_eSPI_User_Setup.zip)

Guerillah, 2020-12-05:
I removed the highlighted part in the source code and it fixed the compiling issue... I'm not sure it should be there anymore. I now have bouncy ball working with only that change.

[image]

Guerillah, 2020-12-05:
I just finished testing the DMA with LVGL and unfortunately, it didn't provide any performance benefit (maybe even a tiny bit of a hit to FPS and processor usage). I am glad I was able to test it though.

Bodmer, 2020-12-06:
v 2.3.51 is the latest and only available with a direct download from Github (not through IDE library maager.

You have changed the right line. Not all RPi dsplays work with DMA, so [this is the line ](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_ESP32.h#L55)I have used.

WacKEDmaN, 2020-12-19:
im having same issues with anything using DMA..
it just errors on compiling.. 
updated to latest github version... now im just getting a white screen on everything it try...(yes i changed User_Setup.h to my need)
back to 2.3.4 and its working again...apart from DMA...

```
sketch\boing_ball.ino.cpp.o:(.literal._Z5setupv+0x24): undefined reference to `TFT_eSPI::initDMA()'
sketch\boing_ball.ino.cpp.o:(.literal._Z4loopv+0x8c): undefined reference to `TFT_eSPI::pushPixelsDMA(unsigned short*, unsigned int)'
sketch\boing_ball.ino.cpp.o: In function `setup()':
E:\WacKEDmaN\Documents\Arduino\libraries\TFT_eSPI\examples\DMA test\boing_ball/boing_ball.ino:66: undefined reference to `TFT_eSPI::initDMA()'
sketch\boing_ball.ino.cpp.o: In function `loop()':
E:\WacKEDmaN\Documents\Arduino\libraries\TFT_eSPI\examples\DMA test\boing_ball/boing_ball.ino:163: undefined reference to `TFT_eSPI::pushPixelsDMA(unsigned short*, unsigned int)'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32 Dev Module.

```

Bodmer, 2020-12-19:
I find Setup27_RPi_ST7796_ESP32.h compiles fine. This is for the MHS-4.0 ST7796 dsplay.

I have [updated this line](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_ESP32.h#L55), so DMA is enabled for all RPI displays. Not all RPi displays can handle DMA, but it is worth a try.

WacKEDmaN, 2020-12-23:
..still having issues.. using RPI_ILI9486_DRIVER and ILI9486_DRIVER.. both work without DMA..
im using one of these https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A) 
..i guess my screen isnt supported :(

Bodmer, 2020-12-23:
RPi displays based on 74HC04, 74HC4040 and 2 x 74HC4094 logic chips are rather picky on command and data timing, so the display is probably getting the data too fast. Thanks for reporting this.

The new ones like the MHS-4.0 inch Display-B type ST7796 are OK with DMA.

WacKEDmaN, 2020-12-23:
yes thats sounds right.. same as pictured here.. https://github.com/Bodmer/TFT_eSPI/blob/master/Tools/RPi_TFT_Connections.png

i swear the other day i was running this at 40mhz spi.. it would give a warning when compiling, and the graphics tests where much faster..  (default 27Mhz worked fine too)..
but now it doesnt like anything above 26.66Mhz.. even with the stable release