Bodmer/TFT_eSPI #842: no matching function for call to 'TFT_eSPI::getTouch(unsigned int*, unsigned int*)'

tonyc770, 2020-11-28:
I get this error message when compiling my program.
The strange this is that your example program works fine and so does the calibration program.

My program prints to the TFT, no problem as well. Seems like I am missing a dependency, or something else, but I can't seem to figure it out.

My loop() function is:
`void loop()
{
  unsigned int touch_X, touch_Y;
  if(tft.getTouch(&touch_X, &touch_Y)){
    Serial.printf("Touched at: %d, %d", touch_X, touch_Y);
  }
   yield();
}

Any ideas???

Thanks...`

Bodmer, 2020-11-28:
The function expects 16 bit unsigned types, so use:
```
uint16_t touch_X, touch_Y;
```
An unsigned int is 32 bits on some processors, hence it is better use of types that are of consistent size.

tonyc770, 2020-11-28:
That fixed the compile, thanks..

But with that loop code running, I get a periodic WatchDog timeout issue again even though I have the yield() function within the loop. If I comment it out, the timeout issue is gone...

Bodmer, 2020-11-28:
So, are you saying this sketch crashes?

```
#include <SPI.h>
#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();

void setup(void) {
  Serial.begin(115200);
  Serial.println("\n\nStarting...");

  tft.init();
}

void loop() {
  uint16_t touch_X, touch_Y;
  if (tft.getTouch(&touch_X, &touch_Y)) {
    Serial.printf("Touched at: %d, %d", touch_X, touch_Y);
  }
  yield();
}
```

tonyc770, 2020-11-28:
The loop function, yes.
My program fetches a weather radar image every 15 minutes and weather data every 30 minutes. A clock updates every minute. Else just the timers ticker.h are running. The code has been working well until I added the touch code.

Bodmer, 2020-11-29:
So... you actually have a complex sketch running with the ticker causing interrupts!
You need to manage the program flow so SPI bus access by the touch function is not interrupted, especially not by TFT access on the same SPI bus!

closed 2020-11-29 by Bodmer