Bodmer/TFT_eSPI #844: Bug in 2.3.41 with ILI_9486 / ESP32_PARALLEL mode?

LineF, 2020-11-29:
Hi,
since 2.3.41 I have display errors.
It looks like only every second line is being drawn.
Going back to 2.3.4 solves the problem.

I'm using your lib on an ESP32 (DevKit V1) in parallel 8 bit mode for an ILI 9486 display.

Can I give you more specific information?

Martin

Bodmer, 2020-11-30:
Should be fixed now in latest Github copy.

Thanks for reporting this.

closed 2020-11-30 by Bodmer

LineF, 2020-11-30:
Hi,

sorry for reopening this case, but the fix didn't resolve the problem. Still the same problem.
[image]

Bodmer, 2020-12-01:
The screen picture tells me you have not updated the library, or have copied in an User_Setup_Select.h old file. [This line was changed](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L119) in "User_Setup_Select.h" make sure you copy has that changed line.

LineF, 2020-12-01:
Yes, I saw that this line was changed.
In my tests each time when I change your library version in platformio.ini before compilation I completely delete the .pio directory.
PlatformIO then must download the current version: when it downloads V2.3.4 the display looks ok, when PIO downloads V2.3.41 then the display looks like above.

My setting in platformio.ini are:

[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino

lib_deps =  bodmer/TFT_eSPI@2.3.4
            Adafruit GFX Library
            Adafruit BusIO
            Wire
            adafruit/Adafruit TouchScreen

build_flags =
  -Os
  -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG
  -DUSER_SETUP_LOADED=1
  -DESP32_PARALLEL=1
  -DILI9486_DRIVER=1
  -DTFT_CS=33
  -DTFT_DC=15
  -DTFT_RST=32
  -DTFT_WR=4
  -DTFT_RD=2
  -DTFT_D0=12
  -DTFT_D1=13
  -DTFT_D2=26
  -DTFT_D3=25
  -DTFT_D4=19
  -DTFT_D5=16
  -DTFT_D6=27
  -DTFT_D7=14
  -DLOAD_GLCD=1
  -DLOAD_FONT2=1
  -DLOAD_FONT4=1

If I replace the line "lib_deps =  bodmer/TFT_eSPI@2.3.4" by "lib_deps =  bodmer/TFT_eSPI" then V2.3.41 is downloaded and I get the problem.

But I will double check this evening that the downloaded lib V2.3.41 will contain the newest version of the changed line.

Thanks so far!
Martin

Bodmer, 2020-12-01:
I have raised the version to 2.3.42 to force a PlatformIO cache flush. That may help, but it can take 24hrs for PlatformIO links to be created.

LineF, 2020-12-01:
Hi,
the raising to 2.3.42 helped.
Now the #ifdef..... line is completely available and the new lib version (2.3.42) works as same as the 2.3.4.

I did a recheck of version 2.3.41 and advised platformio to use this version.
There the line still has the old buggy format (missing the ESP32_PARALLEL entry).
It seems to publish a change in a lib within platformio the lib _must_ be published with a new version number...

Thanks again!
Martin

Bodmer, 2020-12-01:
Thanks for the update.

commit 4bb3712a (2020-11-30): Fix #844