Bodmer/TFT_eSPI #845: Support for 8-bit Parallel interface with touch enabled

boddydan, 2020-11-30:
Hi,
I've been able to use the library to display the examples on my ILI9486 480x320 screen however I cannot get touch support to work.  I've attempted to define TOUCH_CS as my cs pin on my display however receive the following error:
`spi' was not declared in this scope spi.transfer(0xb0);` (I assume this is because I have defined ESP32_PARALLEL.)
I was wondering if the touch mode is supported for my parallel interface or whether I would have to use a different library (or buy a different TFT display)?

Many thanks.

LineF, 2020-11-30:
Hi,
do you have a SPI touch controller or simple 4-wire touch?
My 4-wire touchscreen is working with the adafruit TouchScreen library (and using TFT_eSPI on an ESP32 in 8 bit parallel mode)

Martin

boddydan, 2020-11-30:
> Hi,
> 
> do you have a SPI touch controller or simple 4-wire touch?
> 
> My 4-wire touchscreen is working with the adafruit TouchScreen library (and using TFT_eSPI on an ESP32 in 8 bit parallel mode)
> 
> 
> 
> Martin

Hi Martin,
I'm not sure what you mean by a 4-wire touchscreen.
This is the scren I have:
https://www.amazon.co.uk/Longruner-Arduino-3-5inch-Technical-LSC3A-1/dp/B07K2YCSJD

It uses the RST, CS, RS, WR and RD control lines and has an 8-bit data bus, so 13 wires total.
Thanks,
Dan.

LineF, 2020-11-30:
Hi Dan,
Amazon states that this display has an 12 bit touch controller. But if there would be one, then there should be a CS line for it and I can't see anything like this.
So you have to connect 8 data lines D0-D7, and RD, WR, RS, CS and RST to control the display.

I suspect that you have a simple 4 wire resistive touch screen. This is by default (for RPI) connected to pin 27, 14, 15 and 4 (double use with data lines).

My code looks like this:

#include <Arduino.h>
#include <Adafruit_GFX.h>
#include <TFT_eSPI.h>
#include <TouchScreen.h>
#include "Wire.h"
...
//Define Touch-Parameters
#define MINPRESSURE 50
#define MAXPRESSURE 1000
...
const int coords[] = {935, 150, 130, 980}; // portrait - left, right, top, bottom
TouchScreen ts = TouchScreen(27, 14, 15, 4, 0); // 300 ohm did'nt work - use 0
...

somewhere in the setup() you have to reduce the ADC resolution of the ESP32 from 12 bit to 10bit:
...
  analogReadResolution(10);
...
because the adafruit library can obly handle 10-bit values.

and then somewhere in the code:

  TSPoint p = ts.getPoint();
  pinMode(14, OUTPUT);      //restore shared pins
  pinMode(15, OUTPUT);
  digitalWrite(14, HIGH);   //because TFT control pins
  digitalWrite(15, HIGH);
  Pressed = 0;
  Pressed = (p.z > MINPRESSURE && p.z < MAXPRESSURE);
  if (Pressed) {
    Touch_x = map(p.x, coords[0], coords[1], 0, tft.width()); 
    Touch_y = map(p.y, coords[2], coords[3], 0, tft.height()); 

Hope that helps.
Martin

Bodmer, 2020-12-01:
The TFT_eSPI library does not support touch screens in parallel mode. You can as suggested above use a support library.

closed 2020-12-01 by Bodmer

ramiss, 2024-01-13:
> The TFT_eSPI library does not support touch screens in parallel mode. You can as suggested above use a support library.

@Bodmer - Can you please help me understand why I cannot use the adafruit 4-wire touchscreen.h library alongside the TFT_eSPI?  Surely assigning touch to unused pins would work, no?