Bodmer/TFT_eSPI #846: readPixel issue for TTGO T-Display with ST7789V

ygator, 2020-11-30:
With CGRAM_OFFSET defined readPixel is off 52,40 and display is correct, but if not defined readPixel is correct and display is off 52,40.

Platformio
TFT_eSPI version 2.3.41
build_flags =
    -DTFT_SDA_READ
    -DUSER_SETUP_LOADED
    -DST7789_DRIVER
    -DTFT_WIDTH=135
    -DTFT_HEIGHT=240
    -DCGRAM_OFFSET
    -DTFT_MISO=-1
    -DTFT_MOSI=19
    -DTFT_SCLK=18
    -DTFT_CS=5
    -DTFT_DC=16
    -DTFT_RST=23
    -DTFT_BL=4
    -DTFT_BACKLIGHT_ON=HIGH
    -DLOAD_GLCD
    -DLOAD_FONT2
    -DLOAD_FONT4
    -DLOAD_FONT6
    -DLOAD_FONT7
    -DLOAD_FONT8
    -DLOAD_GFXFF
    -DSMOOTH_FONT
    -DSPI_FREQUENCY=40000000
    -DSPI_READ_FREQUENCY=6000000

ygator, 2020-11-30:
I did this in readPixel and everything works.
The other functions that use the define are setWindow, readAddrWindow, drawPixel, and getSetup.
drawPixel works fine as it is.

#ifdef CGRAM_OFFSET
#ifndef ST7789_DRIVER
  x0+=colstart;
  y0+=rowstart;
#endif
#endif

ygator, 2020-11-30:
Looking more into readPixel I see it does:
#ifdef CGRAM_OFFSET
x0+=colstart;
y0+=rowstart;
#endif

but later calls readAddrWindow which does:
#ifdef CGRAM_OFFSET
  xs += colstart;
  xe += colstart;
  ys += rowstart;
  ye += rowstart;
#endif

So it looks like it is adding the offsets twice.
So the offsets are not needed in readPixel at all.

Bodmer, 2020-12-01:
Fixed, thanks for reporting this.

commit 3b39bf10 (2020-12-01): Fix #846
closed 2020-12-01 by Bodmer