Bodmer/TFT_eSPI #847: ZHONGJINGYUAN ZJY_TFT350_11P_TOUCH

dlbsmith, 2020-12-01:
I am trying to interface with this touch screen display.

It uses a ILI9486L display controller and a XPT2046 touch controller.  I am using an Adafruit ESP32 Huzzah breakout.

The touch controller works fine.

However the display does not.  The screen is white after reboot and then goes dim (black but not as black as when I turn the backlight off), but that is all.

I have selected the following setup:
#include <User_Setups/Setup11_RPi_touch_ILI9486.h> // Setup file configured for ESP32 and RPi TFT with touch

When I run the read user setup sketch I get:

[code]
TFT_eSPI ver = 2.3.4
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9486
Display width  = 320
Display height = 480

MOSI    = GPIO 23
MISO    = GPIO 19
SCK     = GPIO 18
TFT_CS   = GPIO 15
TFT_DC   = GPIO 2
TFT_RST  = GPIO 4
TOUCH_CS = GPIO 22

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 20.00
Touch SPI frequency   = 2.50
[/code]

I have double checked my wiring and the touch does work.

I am not sure what the significance of the RPi is in the setup.  The readme talks about the use of specific chips.  This display does not have those.  Is this significant for the SPI interface?  Have I used the correct setup?

Any guidance would be greatly appreciated.

Bodmer, 2020-12-01:
Post a link to the display, I cannot find it via Google.

Bodmer, 2020-12-01:
I think GPIO 2 is not available on the Adafruit ESP32 Huzzah, what pin have you connected TFT_DC to?

dlbsmith, 2020-12-01:
It is a Adafruit ESP32 Huzzah breakout, not the feather.  Has more pins including GPIO2

https://learn.adafruit.com/huzzah32-esp32-breakout-board

dlbsmith, 2020-12-01:
I can't find any vendor info on the web either.  I do have a datasheet.

[Alt. LCD Datasheet.pdf](https://github.com/Bodmer/TFT_eSPI/files/5619804/Alt.LCD.Datasheet.pdf)

Bodmer, 2020-12-01:
[Comment out line 2 of the setup](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup11_RPi_touch_ILI9486.h#L2), you do not have a Raspberry Pi display board so this option will cause problems!

dlbsmith, 2020-12-01:
Done and it did not make any difference.

I put a print into the setup to check that it did get undefined.

dlbsmith, 2020-12-01:
This setup is using different pins for SPI than the Adafruit standard.  I assume the library overrides the Adafruit standard setup.  Would it be worth while to try changing the SPI pins around to match?

Bodmer, 2020-12-01:
Try the latest library direct from github  version  2.3.41

dlbsmith, 2020-12-01:
Works!!!!!

Thankyou VERY much for the prompt assistance.

closed 2020-12-01 by Bodmer