Bodmer/TFT_eSPI #849: Touch Issue with ST7796s + XPT2046

ElPercha, 2020-12-06:
First things first: thank you so much for this awesome library :)

I having an issue when trying to use this [display](https://www.aliexpress.com/item/4001122632293.html?spm=a2g0s.9042311.0.0.43da63c0tifI5c). Specifically the 4" + Touch (ST7796s + XPT2046)

To summarize:
   - I was able to test the Display and it works OK
   - I was able to test the Touch and it works OK 
   - When trying to use both at the same time NOK
          - Main issue is Touch getting pressed status continuously when calling: tft.getTouch(&ts_x, &ts_y);   
          - I can initialize the Touch not affecting the Display but when I call tft.getTouch the display has also color issues

I tried multiple SPI frequencies combinations for both Display and Touch with same results. 
Also adding delays between calls is not helping.

I'm testing with an ESP32 with the following pinout:
- #define TFT_CS   14
- #define TFT_DC   27
- #define TFT_RST  33
- #define TOUCH_CS 12

The same code and pinout works perfect with a 320*240 (ILI9341_DRIVER + XPT2046) display.
The issue appears when I switch to the bigger physical display and switch the library to ST7796_DRIVER

My guess is that the display is leaving same pin (maybe the SO?) in a state that is affecting the Touch readings. But I'm not sure how to prove or fix this.

Is there a way to modify the ST7796_DRIVER to tristate his outputs (Display Outputs) after each command?
Maybe this is a silly statement but is my best guess :)

Thank you all again to make this library possible.

Bodmer, 2020-12-06:
The ST7796 tristates the SDO (MISO) output. You are getting some odd behaviour, reading the touch controller should not corrupt the display. I suspect this is poor ground, power or a signal integrity problem. Shorten wires, add extra GND signals, check power feed is adequate with no intermittent connections.

ElPercha, 2020-12-07:
Thank you for your reply, the connections seems to be good. I measured 3.25V and it is very stable, everything is in a board, check the pictures...

This is the prototype working with the 2.4" display:
[image]

Both displays have the same connections:
[image]

This is the schematic:
[image]

Just in case I jumped out the 3V3 direct connection in the display, to be sure that the internal regulator is not dropping voltage:
[image]

Unfortunately the issue stills there, see. This is the 4" Display with the touch just initialized:
[image]

This happens when I include the line:
`    pressed = tft.getTouch(&ts_x, &ts_y);   
`
[image]

In your experience, do we need pull-ups in the SPI? is my understanding that not but I'm not sure now. 
Anyway this seems to be software for me because the colors changed and the letters are mirrored!

I know this is very weird :)

Bodmer, 2020-12-07:
Run the Read_User_Setup sketch with the ST7796 to check the compiler is picking up the right pins.

Run the Touch_Calibrate sketch to see if that runs OK, that sketch drives both the display and reads the touch controller.

ElPercha, 2020-12-08:
Yes, I tested with that example yesterday with same results.
I have wired the components in a protoboard to be able to play with the cables individually:

[image]

I added this code to the "TFT_Rainbow480" example:

```
  pressed = tft.getTouch(&ts_x, &ts_y);   
  if (pressed) {
    Serial.print("x = ");
    Serial.print(ts_x);
    Serial.print(" -  y = ");
    Serial.println(ts_y);

    Serial.print("Rotation: ");
    Serial.println(tft.getRotation());
  }
```

It turns out that the Serial Output (MISO) of the ST7796s controller stays down even when CS is high, it suppose to be high impedance. I tested with a diode and pull up and some delays in the code to be completely sure and I confirm.

To summarize, with a 1N4148 diode added everything works as expected:

[image]

If the display is not initialized the MISO is in tristate but after tft.init() it stays down.
I checked the datasheet of the ST7796 but I couldn't find a specific command to tristate the serial output.

Unless you want me to play with some commands to try to fix these with no diode, we should close this issue.

Thank you again for your work and support

Bodmer, 2020-12-08:
Your testing gives interesting (but somewhat conflicting) results. I am not sure the root cause has been identified for a few reasons:
1. The display gets corrupted when the touch controller is accessed, this implies that the TFT is enabled when it chould not be and that commands to the touch controller are getting to the TFT and upsetting it. Display corruption is clearly the case in one of your photos. The question is therefore why is the TFT apparently enabled.
2. The data sheets are not always right but clearly indicate SDO should be going tristate. This again indicates that the chip select is low for some reason. The oddity is that you measure the chip select as being high. Ihave a display like yours without the touch and ststic tests (not dirven by processor) indicate the SDO does go tristate.
3. The touch controller works if the TFT is not accessed, but even the simple Test_Touch_Controller does a tft.init(), so if that was a problem then that sketch should not work either.

The display SDO only needs to be connected if you read from the TFT CGRAM. A diode is not a good solution unless you have a strong pullup on the processor side as the TFT will be unable to drive the processor MISO line high during a read. If you do not need to read the display then a better solution is to remove the connection to the display completely.

I will have a look at my ST7796 board to see if there are any clues to why these odd things are happening. As I am not convinced the root cause has been identified I will leave this open, but it may have to get marked "unresolved" eventually!

Bodmer, 2020-12-08:
I had a look at my display and see a diode D1. This is connected to the TFT_CS line. At first I thought this may be to catch spikes but it appears to be an inline diode, that would be a completely daft idea and is likely to be the root cause of the problems. I will see if I can find a schematic online to confirm. Rather a daft design, if true then shorting out the diode would solve the problems you are having and would explain the behaviour since the chip select would only go high due to parasitic leakage currents which may or may not be the right polarity.
[image]

ldijkman, 2020-12-08:
nice prototype board ;-)
could work for my thermostat

https://youtu.be/qNYT64CaLEI
https://oshwlab.com/l.dijkman/esp32-dev-kit-38-pin-to-spi-touch-tft

Bodmer, 2020-12-08:
The [schematic here](http://www.lcdwiki.com/res/MSP4021/4.0inch_SPI_Schematic.pdf) confirms the diode is inline which is completely daft:
[image]

So short out the diode or fit a low (<50 Ohms) SMT resistor. That will fix the problems you are having. I am happy now as the effect of the diode explains all observations of the fault signature.

ElPercha, 2020-12-09:
I fully agree with your 3 bullets, the results were not clear and let me say, the Diode is a great finding.
Is hard to understand what that Diode is doing, maybe protecting a non 5V tolerant input? It doesn't really matter.
I removed it. I had a hard time removing it :)

[image]
[image]

Unfortunately it didn't work. Same results Touch didn't work when MISO outputs are interconnected. Display OK.

I went 3 steps back:

- I connected the Display CS directly to 3V3. 
- Power up the Display (blank screen of course)
- Measured 3.3V between Display MISO and 3V3 (MISO seems to be shorted to ground)
- Connected a 10K pull up between Display MISO and 3V3
- Measured 3.3V between Display MISO and 3V3 ->MISO is shorted to ground :(
- Measure the other display (I have 2) same thing.

Very disappointed with this detail about the displays since they are very well built and the performance is good. 
A word of warning for potential users that needs to read the CGRAM, do not purchase this model or maybe do not purchase from this Vendor, I don't know.  At least my 2 displays seems to be defective.

**Recommendation: Display MISO should not be connected for this model/Vendor.**

Thank you Bodmer!

julioghigi, 2020-12-09:
Hi!

I also do have this same display, two brand new here, and also can't get touch working with this library, no matter what I do I just get "0,0" from touch, or using a test code  Bodmer gives on another issue I get "x=0 y=0 z~=4000", both display gives me same result.
I was thinking too that my display was defective, but using the arduino example from the LCDWIKI (I wanna this working with ESP32 too):
http://www.lcdwiki.com/4.0inch_SPI_Module_ST7796
that can be download here : http://www.lcdwiki.com/res/Program/Common_SPI/4.0inch/SPI_ST7796S_MSP4020_4021_V1.0/4.0inch_SPI_Module_ST7796S_MSP4020%26MSP4021_V1.0.zip

The "Example_09_touch_pen" for arduino UNO worked really fine with my display, so I know the touch for my display is working, but not with Bodmer library :/

The diode is still on the display.

I am trying to see what LCDWIKI codes do so I can put on Bodmer library, but my coder abilities are not the best :/

ldijkman, 2020-12-09:
maybe you could try my setup 
i have the same screen and 38pin esp
https://github.com/ldijkman/ART-ESP32-Touch-TFT-Thermostat

julioghigi, 2020-12-09:
I tried your setup ldijkman, still no response from Touch, calibration pass by itself, cause it is reading "0,0" from touch.

julioghigi, 2020-12-09:
I am able to get touch info with this libray: https://github.com/PaulStoffregen/XPT2046_Touchscreen

Using the "Touch Test" example I get data about the touch from the serial.

Did the test with ESP32.

julioghigi, 2020-12-09:
I got it working here only with Bodmer library!!

I had just disconnected the MISO wire from the TFT, and just let the T_DO connected to D19, no more SDO (MISO) connected to D19 of ESP32.

Hope this is the same issue you have!

Thanks Bodmer for this awesome library!

Bodmer, 2020-12-09:
These displays have a design fault as described above, a diode is fitted inline with the TFT_CS line, this allows the CS line to be pulled low but only parasitic currents pull it high (aided slightly by diode capacitance on a rising edge).  Since CS tends to stick low the SDO line never tristates and the touch controller driver is too weak to overcome the TFT drie output. I have removed the diode and fitted a link and my display now tristates the SDO output as it should.

Disconnecting the SDO line will help access the touch controller but commands sent to the touch controller still get snet to the TFT corrupting the display. This may appear as extra random pixels. So the correct fix is to modify the board or not use the touch controller (or connect the touch controller to different pins and use a different touch library).

The diode D1 is a tiny silicon diode, it can be removed with a soldering iron. Use a big blob of solder on the iron tip to cover the diode and "wipe" the diode off the pad. Then either bridge the pads with solder, or a wire link, or fit a low (<50) or zero Ohm SMT resistor.
[image]

ElPercha, 2020-12-11:
Thank you all for your help

Just to clarify, my setup also works perfectly fine using TFT_eSPI library for touch and display simultaneously but only if I disconnect MISO of the ST7796s. This is the same with both displays, with and without the diode.

@julioghigi I was not able to compile the example you mentioned for the ESP32, have you tried with other platform? 

On top of the diode issue, I believe that I have to have units from a bad batch where the ST7796s MISO is grounded.
Any of you see a mistake in this test and the mentioned conclusion?

- I connected the Display CS directly to 3V3.
- Power up the Display (blank screen of course)
- Measured 3.3V between Display MISO and 3V3 (MISO seems to be shorted to ground)
- Connected a 10K pull up between Display MISO and 3V3
- Measured 3.3V between Display MISO and 3V3 ->MISO is shorted to ground :(
- Measure the other display (I have 2) same thing.

Bodmer, 2020-12-11:
@ElPercha, I understand that you can access the touch controller and TFT with the diode in place and SDO disconnected. This is not the general case since TFT_CS can still stick low with the diode present, so commands to the touch controller may corrupt the display. I suspect with your display the TFT CS line at the TFT controller side of the diode is drifting high, but this is not the general case (my CS tends to drift low).

Users of the display should therefore note that the general case is:
1. You can leave the diode in place if the TFT controller is the **only** device on the SPI bus (e.g. no Touch controller or SD card connected to the same bus).
2. Replace the diode D1 with a wire link if **any other device** is on the SPI bus, (e.g. touch controller or SD card).

If you are unlucky and have a faulty/damaged TFT where SDO is shorted to GND then leave the TFT SDO pin unconnected. However bear in mind that a shorted SDO also increases the consumption of the display when the ST7796 chip tries try to drive the pin high.

closed 2020-12-11 by Bodmer