Bodmer/TFT_eSPI #851: Time in millis since last touch

ldijkman, 2020-12-08:
Hello

this is not the place to ask questions i think => is there a forum about tft espi?
is somewhere a manual of tft espi
my question is 
is there something like => Time in millis since last touch
would like to animate the screen bigger hide buttons if not touched for a certain time

greet luberth
https://www.youtube.com/watch?v=qNYT64CaLEI
https://oshwlab.com/l.dijkman/esp32-dev-kit-38-pin-to-spi-touch-tft

ldijkman, 2020-12-08:
ok saw this in another post
pressed = tft.getTouch(&ts_x, &ts_y);   
 if (pressed) {
get the time pressed
}
maybe that will work

Bodmer, 2020-12-08:
Hi, There is no forum and a user manual is on my to do list but I usually find something more interesting to do!

There is no function for the last touch time, so as you conclude this has to be done in the sketch. It would be possible to use the touch controller interrupt line for this but this is not used by the library.

The touch feature of the library will be deprecated eventually and a seperate library recommended so that different touch controllers can be supported.

closed 2020-12-08 by Bodmer

ldijkman, 2020-12-08:
Thank you for the info 
Would be nice if the keywords list give a bit more info about parameters
keywordslist is drawRoundRect
drawRoundRect(10, 10, 300, 140, 8, BLUE);

https://github.com/Bodmer/TFT_eSPI/blob/master/keywords.txt

Bodmer, 2020-12-08:
The keywords list is for the Arduino IDE and identifies the ones to colour highlight in the editor. The list here:
https://github.com/Bodmer/User_Manual_TFT_eSPI/blob/master/TFT_eSPI.cpp.All.pdf
 was kindly put together by a user but is missing the latest additions.

ldijkman, 2020-12-08:
ok thank you verry much
that makes life easier
i will print it out
thanks

Bodmer, 2020-12-08:
I have started a new "[Discussions](https://github.com/Bodmer/TFT_eSPI/discussions)" tab that may be of interest.