Bodmer/TFT_eSPI #857: Sprite rotation without immediately push

thor171, 2020-12-11:
Hello!
Currently I am working on a project, where I need to get the framerate as high as possible. In order to achieve this, it would be great, if the 16bpp sprites could be rotated without immediately pushing them to the tft. So I could compute their rotation on one CPU core while transmitting the prerendered image using the other core.

Bodmer, 2020-12-11:
The transformation sampling process means this can only be achieved via an image buffer. You can push a rotated copy of a Sprite to another sprite. Function prototype is:
```
           // Push a rotated copy of Sprite to another different Sprite with optional transparent colour
  bool     pushRotated(TFT_eSprite *spr, int16_t angle, int32_t transp = -1); 
```

thor171, 2020-12-11:
Ohh, I didn't notice. Thank you very much!

Your library is great! I really like it :+1:

closed 2020-12-16 by Bodmer