Bodmer/TFT_eSPI #866: ILI9486 8bit Touch, not working.

Trackhe, 2020-12-17:
I use 

https://www.amazon.de/Niunion-Touch-LCD-Bildschirmmodul-480x320-Kompatibel/dp/B08KG51VLW

```
#define ILI9486_DRIVER

#define TFT_PARALLEL_8_BIT

// The ESP32 and TFT the pins used for testing are:
#define TFT_CS   33  // Chip select control pin (library pulls permanently low
#define TFT_DC   15  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  32  // Reset pin, toggles on startup

#define TFT_WR    4  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD    2  // Read strobe control pin

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits.
#define TFT_D2   26  // Pins can be randomly assigned, this does not affect
#define TFT_D3   25  // TFT screen update performance.
#define TFT_D4   17
#define TFT_D5   16
#define TFT_D6   27
#define TFT_D7   14
```

any idea why it is not working?

Trackhe, 2020-12-17:
I get 
```
19:06:25.262 -> 
19:06:25.262 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x3b (SPI_FAST_FLASH_BOOT)
19:06:25.299 -> flash read err, 1000
19:06:25.299 -> Falling back to built-in command interpreter.
19:06:25.299 -> OK
19:06:25.299 -> >ets Jun  8 2016 00:22:57
19:06:25.634 -> 
```

I f i use D0 on GPIO12
i change D0 to GPIO15 and the esp boots and run the programm but the display is only flickering if i run a benchmark

Bodmer, 2020-12-17:
The library does not support analogue interface touch screens.

Bodmer, 2020-12-17:
You should be able to get the screen display working and then use a third party library for touch. The pinout you are using in the setup is correct. If you are using an UNO style ESP32 board then make sure you add the wires [as shown here](https://github.com/Bodmer/TFT_eSPI/blob/master/Tools/ESP32%20UNO%20board%20mod/ESP32%20UNO%20board%20mod.jpg).

Trackhe, 2020-12-18:
Thank you for your reply. 
"The library does not support touchscreens with an analog interface." I thought so and I'm fine with that. But I use [Board] as my board. how i can archiv this wirering here? (https://www.banggood.com/de/ESP32-Test-Board-Small-Batch-Burn-Fixture-Min-System-Development-Board-For-ESP-WROOM-32-ESP-32S-p-1163923.html?utm_source=googleshopping&utm_medium=cpc_organic&gmcCountry=DE&utm_content=minha&utm_campaign=minha-de-de-pc&currency=EUR&cur_warehouse=CN&createTmp=1&utm_source=googleshopping&utm_medium=cpc_bgs&utm_content=frank&utm_campaign=frank-pla-de-all-0513&ad_id=435242839872&gclid=CjwKCAiAoOz-BRBdEiwAyuvA69uoMW6_HZ5aX1RLo2IodLfPGOnRde0lFTpdfApBrk--HRAjnFZ8SRoC4X8QAvD_BwE)

Trackhe, 2020-12-18:
I connected these Pins but same resultonly white screen that flickers. i reset the pins to standard -> but then the esp reboots and reboots and so on

```
#define TFT_CS   33  // Chip select control pin (library pulls permanently low
#define TFT_DC   15  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  32  // Reset pin, toggles on startup

#define TFT_WR    4  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD    2  // Read strobe control pin

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits.
#define TFT_D2   26  // Pins can be randomly assigned, this does not affect
#define TFT_D3   25  // TFT screen update performance.
#define TFT_D4   17
#define TFT_D5   16
#define TFT_D6   27
#define TFT_D7   14
`
``

Bodmer, 2020-12-19:
Make sure you have a good GND connection between ESP32 and display, use at least 2 GND wires.
Add a 1K Ohm pull down on the GPIO 12 to GND as it appears when the display is connected that pin goes high. You could also try connecting TFT_RD from the display direct to +3.3V to make sure the board buffers are not enabled and driving the data bus pins high. If you want to read from the display in future you will beed to connect it again to the GPIO pin, in this case you may need a 1K Ohm pullup to +3.3V to keep the display disabled during boot.

Bodmer, 2020-12-19:
[This link](https://randomnerdtutorials.com/esp32-pinout-reference-gpios/) has a table which showns IO12 has a dual function at boot.

Trackhe, 2020-12-20:
good GND connection between ESP32 and display # check

1K Ohm pull down on the GPIO 12 to GND # No Notable change.
TFT_RD from the display direct to +3.3V # No Notable change.

due to private matters, testing with the uno r32 is unfortunately delayed until tomorrow.

Trackhe, 2020-12-20:
So after long after research i found an lib that work with the display on a Mega 2560
http://www.lcdwiki.com/3.5inch_Arduino_Display-UNO#Program_Download
[Demo_Arduino.zip](https://github.com/Bodmer/TFT_eSPI/files/5720872/Demo_Arduino.zip)

Can we port it to TFT_eSPI?

Trackhe, 2020-12-22:
After Killing my DP i test it with another one and now i guess it works.

closed 2020-12-22 by Trackhe