Bodmer/TFT_eSPI #868: error: undefined reference to `TFT_eSPI::getTouch(unsigned short*, unsigned short*, unsigned short)'

lex007112, 2020-12-17:
I can not compile the firmware

> Executing task in folder esp32_thermostat-master: C:\Users\le_x_\.platformio\penv\Scripts\pio.exe run <

Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2.0.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.201016 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 41 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit SHT31 Library> 2.0.0
|   |-- <Adafruit BusIO> 1.7.1
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|-- <JPEGDecoder> 1.8.1
|   |-- <FS> 1.0
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <SD(esp32)> 1.0.5
|   |   |-- <FS> 1.0
|   |   |-- <SPI> 1.0
|-- <Ticker> 3.2.0
|-- <Blynk> 0.6.7
|   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <SD(esp32)> 1.0.5
|   |   |-- <FS> 1.0
|   |   |-- <SPI> 1.0
|-- <TFT_eSPI> 2.3.52
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <FS> 1.0
|   |-- <SPI> 1.0
|-- <ArduinoJson> 6.17.2
|-- <Adafruit GFX Library> 1.10.4
|   |-- <Adafruit BusIO> 1.7.1
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <Adafruit ILI9341> 1.5.6
|   |-- <Adafruit GFX Library> 1.10.4
|   |   |-- <Adafruit BusIO> 1.7.1
|   |   |   |-- <Wire> 1.0.1
|   |   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|-- <Display>
|   |-- <EEPROM> 1.0.3
|   |-- <FS> 1.0
|   |-- <TFT_eSPI> 2.3.52
|   |   |-- <SPIFFS> 1.0
|   |   |   |-- <FS> 1.0
|   |   |-- <FS> 1.0
|   |   |-- <SPI> 1.0
|   |-- <JPEGDecoder> 1.8.1
|   |   |-- <FS> 1.0
|   |   |-- <SPIFFS> 1.0
|   |   |   |-- <FS> 1.0
|   |   |-- <SD(esp32)> 1.0.5
|   |   |   |-- <FS> 1.0
|   |   |   |-- <SPI> 1.0
|   |-- <MeteoData>
|   |   |-- <Adafruit SHT31 Library> 2.0.0
|   |   |   |-- <Adafruit BusIO> 1.7.1
|   |   |   |   |-- <Wire> 1.0.1
|   |   |   |   |-- <SPI> 1.0
|   |   |   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|   |-- <WiFi> 1.0
|-- <EEPROM> 1.0.3
|-- <InternetConnection>
|   |-- <ArduinoOTA> 1.0
|   |   |-- <Update> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <ESPmDNS> 1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <Display>
|   |   |-- <EEPROM> 1.0.3
|   |   |-- <FS> 1.0
|   |   |-- <TFT_eSPI> 2.3.52
|   |   |   |-- <SPIFFS> 1.0
|   |   |   |   |-- <FS> 1.0
|   |   |   |-- <FS> 1.0
|   |   |   |-- <SPI> 1.0
|   |   |-- <JPEGDecoder> 1.8.1
|   |   |   |-- <FS> 1.0
|   |   |   |-- <SPIFFS> 1.0
|   |   |   |   |-- <FS> 1.0
|   |   |   |-- <SD(esp32)> 1.0.5
|   |   |   |   |-- <FS> 1.0
|   |   |   |   |-- <SPI> 1.0
|   |   |-- <MeteoData>
|   |   |   |-- <Adafruit SHT31 Library> 2.0.0
|   |   |   |   |-- <Adafruit BusIO> 1.7.1
|   |   |   |   |   |-- <Wire> 1.0.1
|   |   |   |   |   |-- <SPI> 1.0
|   |   |   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <EEPROM> 1.0.3
|   |-- <HTTPClient> 1.2
|   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <MeteoData>
|   |   |-- <Adafruit SHT31 Library> 2.0.0
|   |   |   |-- <Adafruit BusIO> 1.7.1
|   |   |   |   |-- <Wire> 1.0.1
|   |   |   |   |-- <SPI> 1.0
|   |   |   |-- <Wire> 1.0.1
|   |-- <Thermostat>
|   |   |-- <Display>
|   |   |   |-- <EEPROM> 1.0.3
|   |   |   |-- <FS> 1.0
|   |   |   |-- <TFT_eSPI> 2.3.52
|   |   |   |   |-- <SPIFFS> 1.0
|   |   |   |   |   |-- <FS> 1.0
|   |   |   |   |-- <FS> 1.0
|   |   |   |   |-- <SPI> 1.0
|   |   |   |-- <JPEGDecoder> 1.8.1
|   |   |   |   |-- <FS> 1.0
|   |   |   |   |-- <SPIFFS> 1.0
|   |   |   |   |   |-- <FS> 1.0
|   |   |   |   |-- <SD(esp32)> 1.0.5
|   |   |   |   |   |-- <FS> 1.0
|   |   |   |   |   |-- <SPI> 1.0
|   |   |   |-- <MeteoData>
|   |   |   |   |-- <Adafruit SHT31 Library> 2.0.0
|   |   |   |   |   |-- <Adafruit BusIO> 1.7.1
|   |   |   |   |   |   |-- <Wire> 1.0.1
|   |   |   |   |   |   |-- <SPI> 1.0
|   |   |   |   |   |-- <Wire> 1.0.1
|   |   |   |-- <SPI> 1.0
|   |   |   |-- <WiFi> 1.0
|   |   |-- <EEPROM> 1.0.3
|   |   |-- <MeteoData>
|   |   |   |-- <Adafruit SHT31 Library> 2.0.0
|   |   |   |   |-- <Adafruit BusIO> 1.7.1
|   |   |   |   |   |-- <Wire> 1.0.1
|   |   |   |   |   |-- <SPI> 1.0
|   |   |   |   |-- <Wire> 1.0.1
|   |-- <WiFi> 1.0
|   |-- <Blynk> 0.6.7
|   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|   |   |-- <SD(esp32)> 1.0.5
|   |   |   |-- <FS> 1.0
|   |   |   |-- <SPI> 1.0
|-- <MeteoData>
|   |-- <Adafruit SHT31 Library> 2.0.0
|   |   |-- <Adafruit BusIO> 1.7.1
|   |   |   |-- <Wire> 1.0.1
|   |   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0.1
|-- <Thermostat>
|   |-- <Display>
|   |   |-- <EEPROM> 1.0.3
|   |   |-- <FS> 1.0
|   |   |-- <TFT_eSPI> 2.3.52
|   |   |   |-- <SPIFFS> 1.0
|   |   |   |   |-- <FS> 1.0
|   |   |   |-- <FS> 1.0
|   |   |   |-- <SPI> 1.0
|   |   |-- <JPEGDecoder> 1.8.1
|   |   |   |-- <FS> 1.0
|   |   |   |-- <SPIFFS> 1.0
|   |   |   |   |-- <FS> 1.0
|   |   |   |-- <SD(esp32)> 1.0.5
|   |   |   |   |-- <FS> 1.0
|   |   |   |   |-- <SPI> 1.0
|   |   |-- <MeteoData>
|   |   |   |-- <Adafruit SHT31 Library> 2.0.0
|   |   |   |   |-- <Adafruit BusIO> 1.7.1
|   |   |   |   |   |-- <Wire> 1.0.1
|   |   |   |   |   |-- <SPI> 1.0
|   |   |   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <EEPROM> 1.0.3
|   |-- <MeteoData>
|   |   |-- <Adafruit SHT31 Library> 2.0.0
|   |   |   |-- <Adafruit BusIO> 1.7.1
|   |   |   |   |-- <Wire> 1.0.1
|   |   |   |   |-- <SPI> 1.0
|   |   |   |-- <Wire> 1.0.1
Building in release mode
Compiling .pio\build\esp32dev\src\main.cpp.o
Compiling .pio\build\esp32dev\liba0a\Display\Display.cpp.o
Compiling .pio\build\esp32dev\liba0a\Display\GfxUi.cpp.o
Compiling .pio\build\esp32dev\liba0a\Display\JpegUtils.cpp.o
Compiling .pio\build\esp32dev\lib5b4\Thermostat\Thermostat.cpp.o
Compiling .pio\build\esp32dev\libf34\InternetConnection\InternetConnection.cpp.o
Archiving .pio\build\esp32dev\liba0a\libDisplay.a
Archiving .pio\build\esp32dev\lib5b4\libThermostat.a
Archiving .pio\build\esp32dev\libf34\libInternetConnection.a
Linking .pio\build\esp32dev\firmware.elf
.pio\build\esp32dev\liba0a\libDisplay.a(Display.cpp.o):(.literal._ZN7Display19checkDisplayClickedEv+0xc): undefined reference to `TFT_eSPI::getTouch(unsigned short*, unsigned short*, unsigned short)'
.pio\build\esp32dev\liba0a\libDisplay.a(Display.cpp.o): In function `Display::checkDisplayClicked()':
E:\firmware_vscode\esp32_thermostat-master/lib\Display/Display.cpp:287: undefined reference to `TFT_eSPI::getTouch(unsigned short*, unsigned short*, unsigned short)'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
================================================================================================== [FAILED] Took 19.66 seconds ==================================================================================================

Bodmer, 2020-12-17:
You need to have a touch screen based on XPT2046 and define the touch chip select pin in the setup file.

lex007112, 2020-12-17:
everything was solved in a strange way, by adding the tft_hspi library.  The board was requested, but the project did not work ... let's figure it out further ..

closed 2020-12-17 by lex007112