Bodmer/TFT_eSPI #871: ILI9486 Parallel and ESP32 NodeMCU blank screen and weird stuff

osmaga2012, 2020-12-18:
Hello everyone. I have the following screen, which I bought from amazon

https://www.amazon.es/gp/product/B07N4KXP5X/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&th=1

and an ESP32 NodeMCU module also purchased from amazon,

https://www.az-delivery.de/en/products/esp32-developmentboard

I have tried to make it work but there is no way. I have connected the screen with the following configuration

// The ESP32 and TFT the pins used for testing are:
#define TFT_CS 5 // 33 // Chip select control pin (library pulls permanently low)
#define TFT_DC 15 // 15 // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST 18 // 32 // Reset pin, toggles on startup

#define TFT_WR 4 // 4 // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD 2 // 2 // Read strobe control pin

#define TFT_D0 12 // Must use pins in the range 0-31 for the data bus
#define TFT_D1 13 // so a single register write sets / clears all bits.
#define TFT_D2 26 // Pins can be randomly assigned, this does not affect
#define TFT_D3 25 // TFT screen update performance.
#define TFT_D4 14 // 17
#define TFT_D5 27 // 16
#define TFT_D6 33 // 27
#define TFT_D7 32 // 14

I have also tried the original pin configuration but it still does the same. White screen.

If I connect it to an Arduino UNO R3 it seems to work but not with the expected result, as shown in the image.

[image]
[image]

Previously, I switched screens for the same problem and I already doubt that it is a problem of the two screens.

Another question I have is about voltages. The screen has two pins one of 3V3 and another of 5V, which one to use? My ESP32 has two pins. Also, the screen has two ground connections ... Which one to use?

Thanks in advance

Bodmer, 2020-12-18:
Note comment on this line:

```
#define TFT_D0 12 // Must use pins in the range 0-31 for the data bus
```
Note your modified setup is:
```
#define TFT_D6 33 // 27
```
So you have not complieditht the pin allocation needs.

Typically these displays run from 5V only, and do not use the 3.3V pin.
Connect both grounds to the ESP32, this will aid signal integrity.

osmaga2012, 2020-12-19:
Thanks for answering!

My connection is now like this

// The ESP32 and TFT the pins used for testing are:
#define TFT_CS 5 // 33 // Chip select control pin (library pulls permanently low)
#define TFT_DC 15 // 15 // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST 18 // 32 // Reset pin, toggles on startup

#define TFT_WR 4 // 4 // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD 2 // 2 // Read strobe control pin

#define TFT_D0 12 // Must use pins in the range 0-31 for the data bus
#define TFT_D1 13 // so a single register write sets / clears all bits.
#define TFT_D2 26 // Pins can be randomly assigned, this does not affect
#define TFT_D3 25 // TFT screen update performance.
#define TFT_D4 14 // 17
#define TFT_D5 27 // 16
#define TFT_D6 23 // 33 // 27
#define TFT_D7 22 // 32 // 14

[image]

I have connected both GND and 5V ...

Thank you.

Now with this configuration the screen turns on, but it appears as in the attached image.

Trackhe, 2020-12-19:
I try it too but my screen are always white

osmaga2012, 2020-12-19:
I've read that patch cables can influence performance, is that true?

Bodmer, 2020-12-19:
@osmaga2012 Make sure your connections from the ESP32 to the display are as short as possible (less than100mm). I have exactly the same display and it works fine on a UNO style ESP32 with the default connections. Direct wiring can be problematic due to signal integrity issues.

@Trackhe If GPIO12 is high at boot time, then boot will fail. Try disconnecting that line and see if it boots. I assume you have an identical display to @osmaga2012?

Things are much easier if you use an UNO style ESP32 board [with these mods](https://github.com/Bodmer/TFT_eSPI/blob/master/Tools/ESP32%20UNO%20board%20mod/ESP32%20UNO%20board%20mod.jpg)!

osmaga2012, 2020-12-19:
thanks @Bodmer  for the great work you do !!!

To have where I get shorter cables ... mine are 20mm ... I have several protoboards but it is impossible for me to prick it ... 

Still ... is it normal for my screen to look like this even when using it on an Arduino UNO R3?

Trackhe, 2020-12-19:
@Bodmer this is my dp https://www.amazon.de/Niunion-Touch-LCD-Bildschirmmodul-480x320-Kompatibel/dp/B08KG51VLW
and if i disconnecting the D= the esp starts

Bodmer, 2020-12-19:
@osmaga2012 No that is not normal for an UNO. It looks like the library is not configuring the display correctly. Check you are using the right display driver.

@Trackhe and @osmaga2012  Both the advert you provide links to indicates they might supply one of two different display drivers (viz "Controlador IC: ILI9486 / ILI9488L" "Fahrt IC: ILI9486/ILI9488L ") so you may have an ILI9488 type and thus will need to select a different driver in the TFT_eSPI setup file ( i.e. ILI9488_DRIVER).

Trackhe, 2020-12-19:
@osmaga2012 i have already test it with the ILI9488 but same result.

Bodmer, 2020-12-19:
I've just downloaded a fresh copy of version 2.3.4, selected "Setup13_ILI9481_Parallel.h" as the setup file and changed ILI9486 so it reads:
```
#define ESP32_PARALLEL

#define ILI9486_DRIVER

// ESP32 pins used for UNO format board
#define TFT_CS   33  // Chip select control pin
#define TFT_DC   15  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  32  // Reset pin

#define TFT_WR   4  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD   2

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits
#define TFT_D2   26
#define TFT_D3   25
#define TFT_D4   17
#define TFT_D5   16
#define TFT_D6   27
#define TFT_D7   14

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT
```

Note the : #define ESP32_PARALLEL
Sketches compile, upload and work fine on my display (the display back PCB side visualy looks identical to your advert pictures).
I amusing the UNO style ESP32 as it eliminates wiring errors.

I have run out of ideas, you could try running the diagnostic sketch "Read_User_Setup". Check the serial output carefully. This is my output with that skectch, this verifies the compiler has picked up the right setup file and a setup mistake has not been made:
```
TFT_eSPI ver = 2.3.4
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = Parallel
Display driver = 9486
Display width  = 320
Display height = 480

TFT_CS   = GPIO 33
TFT_DC   = GPIO 15
TFT_RST  = GPIO 32
TFT_WR   = GPIO 4
TFT_RD   = GPIO 2

TFT_D0   = GPIO 12
TFT_D1   = GPIO 13
TFT_D2   = GPIO 26
TFT_D3   = GPIO 25
TFT_D4   = GPIO 17
TFT_D5   = GPIO 16
TFT_D6   = GPIO 27
TFT_D7   = GPIO 14

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

```

Trackhe, 2020-12-19:
I had make this yesterday 
```
16:54:03.365 -> TFT_eSPI ver = 2.3.51
16:54:03.365 -> Processor    = ESP32
16:54:03.365 -> Frequency    = 240MHz
16:54:03.365 -> Transactions = Yes
16:54:03.365 -> Interface    = Parallel
16:54:03.365 -> Display driver = 9486
16:54:03.365 -> Display width  = 320
16:54:03.365 -> Display height = 480
16:54:03.365 -> 
16:54:03.365 -> TFT_CS   = GPIO 33
16:54:03.365 -> TFT_DC   = GPIO 18
16:54:03.365 -> TFT_RST  = GPIO 32
16:54:03.365 -> TFT_WR   = GPIO 4
16:54:03.365 -> TFT_RD   = GPIO 2
16:54:03.365 -> 
16:54:03.365 -> TFT_D0   = GPIO 12
16:54:03.365 -> TFT_D1   = GPIO 13
16:54:03.402 -> TFT_D2   = GPIO 26
16:54:03.402 -> TFT_D3   = GPIO 25
16:54:03.402 -> TFT_D4   = GPIO 17
16:54:03.402 -> TFT_D5   = GPIO 16
16:54:03.402 -> TFT_D6   = GPIO 27
16:54:03.402 -> TFT_D7   = GPIO 14
16:54:03.402 -> 
16:54:03.402 -> Font GLCD   loaded
16:54:03.402 -> Font 2      loaded
16:54:03.402 -> Font 4      loaded
16:54:03.402 -> Font 6      loaded
16:54:03.402 -> Font 7      loaded
16:54:03.402 -> Font 8      loaded
16:54:03.402 -> Smooth font enabled
```

i guess i can test it next day with a uno from a friend.

osmaga2012, 2020-12-19:
> @osmaga2012 No that is not normal for an UNO. It looks like the library is not configuring the display correctly. Check you are using the right display driver.
>
> @Trackhe and @osmaga2012 Both the [...]

Thank you. I will review and test it again. 

I was asking because the previous one I returned for the same reason and I thought it was broken ...

Bodmer, 2020-12-23:
@Trackhe appears to have got it working. Not sure how...

Trackhe, 2020-12-23:
I used a mega and uno for testing the display with another lib. Then i found out my display was broken. Then with the exact same but other display it got worked. Then i test it with the wemos r32 and now it is Working with the benchmark example. So i test it now with my own Programm but this was failing because the touch is not existing. I comment the touch part out and it worked but if lvgl loaded i get not a normal picture.

Bodmer, 2021-01-06:
I am not able to replicate the problem and the display on my setup is running fine, so unfortunately I am not able to advise further. If you do get it working then do post back.

closed 2021-01-06 by Bodmer