Bodmer/TFT_eSPI #872: esp32 TTGO T-Display blank screen

MartinNohr, 2020-12-18:
I was using version 2.2.23, working fine. I switched to the latest code here, 2.2.52. Now all I get is a blank screen.
I've attached the files I modified. That TTGO uses ST7789 driver.

[TFT_eSPInew.zip](https://github.com/Bodmer/TFT_eSPI/files/5716577/TFT_eSPInew.zip)

Bodmer, 2020-12-18:
You have not eddited the setup files correctly.  The T display is suported, replace your modified versions of User_Setup.h and User_Setup_Select.h with the orginal library files, then comment out (add // to line start) [this line](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L22). And uncomment (remove //) [this line](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L53), then recompile and upload.

MartinNohr, 2020-12-18:
Wow, thanks for the fast response. I figured I did something wrong. I'll try again later today.
BTW, your library is incredibly faster than the adafruit one. It is very much appreciated.

MartinNohr, 2020-12-18:
So User_Setup.h is not modified, just the 2 lines in User_Setup_Select.h. I tried this and it worked perfectly. Thanks again for a great library.

closed 2020-12-18 by Bodmer