Bodmer/TFT_eSPI #873: tft.readPixel() not working for TTGO T-Display ST7789V

ThatMantis, 2020-12-18:
Hi, I have a TTGO T-Display ESP32 board that includes a ST7789V 1.14 IPS screen ( https://tinyurl.com/y8lfrav8 ). I have already commented the "#include <User_Setup.h> " and uncommented the "#include <User_Setups/Setup25_TTGO_T_Display.h>" in "User_Setup_Select.h". While most of the functions appear to work fine, the tft.readPixel() function does not appear to be working for me, and just prints "0" in the serial monitor when I try to read it. 

I have also tried adding in the "#define TFT_SDA_READ " line in the "Setup25_TTGO_T_Display.h" file, but tft.readPixel() still fails to work. 

I have already updated the arduino library to version 2.3.4, please advise! Thanks!

lewisxhe, 2020-12-18:
Most TFT screens do not support reading, and there is no MISO, as is the case with TDisplay.  So this is not supported by hardware and has nothing to do with this library!

ThatMantis, 2020-12-19:
Yea but I came across a few threads such as https://github.com/Bodmer/TFT_eSPI/issues/248 which uses the ST7789V without MISO, and threads such as https://forum.arduino.cc/index.php?topic=673549.15 which somehow manages to enable _tft.readPixel()_ with the T-Display board, so I was wondering if perhaps I might have done something wrong or missed something out in the new edition of this library.

Bodmer, 2020-12-19:
The library supports biderectional SPI on ESP32 only and this works OK on my TTGO T v1.1 display.

This is the setup I use:
```
#define ST7789_DRIVER

#define TFT_SDA_READ

#define TFT_WIDTH  135
#define TFT_HEIGHT 240

#define CGRAM_OFFSET      // Library will add offsets required

#define TFT_MISO    -1
#define TFT_MOSI            19
#define TFT_SCLK            18
#define TFT_CS              5
#define TFT_DC              16
#define TFT_RST             23

#define TFT_BL          4  // Display backlight control pin

#define TFT_BACKLIGHT_ON HIGH  // HIGH or LOW are options

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

#define SMOOTH_FONT

#define SPI_FREQUENCY  40000000

#define SPI_READ_FREQUENCY  6000000 // 6 MHz is the maximum SPI read speed for the ST7789V
```

This code:
```
  tft.drawPixel(30,30,TFT_RED);
  Serial.println(tft.readPixel(30,30),HEX);
```
Should print F800

ThatMantis, 2020-12-21:
Hmm that is weird, I am still getting an output of '0'.
This is the code in the _Setup25_TTGO_T_Display.h_ setup file  i am using:
`// Setup for the TTGO T4 ("Bitcoin Tracker") ESP32 board with 2.2" ILI9341 display

// See SetupX_Template.h for all options available

#define ST7789_DRIVER
#define TFT_SDA_READ   

#define TFT_WIDTH  135
#define TFT_HEIGHT 240

#define CGRAM_OFFSET      // Library will add offsets required

#define TFT_MISO -1

#define TFT_MOSI            19
#define TFT_SCLK            18
#define TFT_CS              5
#define TFT_DC              16
#define TFT_RST             23

#define TFT_BL          4  // Display backlight control pin

#define TFT_BACKLIGHT_ON HIGH  // HIGH or LOW are options

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

#define SMOOTH_FONT

//#define SPI_FREQUENCY  27000000
  #define SPI_FREQUENCY  40000000   // Maximum for ILI9341

#define SPI_READ_FREQUENCY  6000000 // 6 MHz is the maximum SPI read speed for the ST7789V
`
and this is my code in Arduino:
`#include <TFT_eSPI.h> // Hardware-specific library
#include <SPI.h>

TFT_eSPI tft = TFT_eSPI(); // Invoke custom library

void setup() {
  Serial.begin(115200); // Set to a high rate for fast image transfer to a PC
  tft.init();
  tft.fillScreen(TFT_BLACK);
  tft.drawPixel(30,30,TFT_RED);
  Serial.println(tft.readPixel(30,30),HEX);
}

void loop() {
Serial.println(tft.readPixel(30,30),HEX);
 }`

Please let me know if you think there is an error in my code or something! Otherwise I might need to get a new board haiz.

ThatMantis, 2020-12-21:
ok this is weird, I continued messing around with the board afterwards and tried this code:
```
#include <TFT_eSPI.h> // Hardware-specific library
#include <SPI.h>

TFT_eSPI tft = TFT_eSPI(); 
void setup() {
  Serial.begin(115200);
  tft.init();
  tft.fillScreen(TFT_BLACK);
  tft.fillRect(10,10,96,128, TFT_BLUE);       //  <----- Test color
  Serial.println(tft.readPixel(30,30),HEX);  // <----- Try to read color
  tft.drawPixel(30,30,TFT_RED);
  Serial.println(tft.readPixel(30,30),HEX);
}

void loop() {
}
```
and it managed to print "1F" for both Serial.println lines, I am guessing the pixel indexing in my TTGO T-Display board is off? Any ideas how to correct that?

Bodmer, 2020-12-21:
Try adding a setRotation() after the init, that should set the offsets. If that works then there is a bug that needs fixing!

ThatMantis, 2020-12-21:
Hmm that does not seem to work, anyway i tried to print out all the pixel values of the screen using the code below:
```
#include <TFT_eSPI.h> // Hardware-specific library
#include <SPI.h>

TFT_eSPI tft = TFT_eSPI(); 
void setup() {
  Serial.begin(115200);
  tft.init();
  tft.setRotation(0);
  tft.fillScreen(TFT_BLACK);
  int w = tft.width();
  int h = tft.height();
  Serial.println(w);
  Serial.println(h);
  for (int y=0; y<h; y++) {
    Serial.println();
    Serial.println();
    for (int x=0; x<w; x++) {
      Serial.print(",");
      Serial.print(tft.readPixel(x,y),HEX); 
}}
}

void loop() {
}

```
i was expecting to get a 135x240 series of "0," since the screen is black, however i got a print out on the serial monitor that looks like this at the start:
[image]
and this at the end:
[image]

ps. they are zoomed out, but it is clear that only the first few indexes are "0", and the rest are not

Bodmer, 2020-12-22:
The ST7789 is deigned for a 240x320 display so typically the 135x240 is mapped near the middle of that area. This mean the offsets are needed. I probably made a mistake when adding the viewport feature that does not handle the offsets correctly.

I will load 2.3.4 and test tomorrow.

Bodmer, 2020-12-22:
I have found the problem. The SPI registers need to be updated when the SDA mode is invoked.

I will be able to update the master tomorrow, you will then have to download a copy direct from Github as it will not be released for a while and thus will not be available using the IDE library menu.  I will post again when the changed have been made, tested and uploaded.

closed 2020-12-23 by Bodmer

ThatMantis, 2020-12-29:
Thanks! It works now!

Bodmer, 2020-12-29:
Great, thanks for letting me know.

electronicsguy, 2025-06-22:
@Bodmer 

readPixel() is not reporting any value other than '0' for ST7735 on ESP32S3. 

Code as below:

```
#include <TFT_eSPI.h> // Hardware-specific library
#include <SPI.h>

TFT_eSPI tft = TFT_eSPI();

void setup() {
  Serial.begin(115200);
  tft.init();
  tft.setRotation(0);
  tft.fillScreen(TFT_RED);

  int w = tft.width();
  int h = tft.height();

  Serial.println();
  Serial.println(w);
  Serial.println(h);

  for (int y=0; y<h; y++) {
    Serial.println();
    Serial.print("y:");
    Serial.println(y);
    
    for (int x=0; x<w; x++) {
      Serial.print(",");
      Serial.print(tft.readPixel(x,y),HEX); 
    }
  }

  Serial.print("\ncolor 1: ");
  Serial.println(tft.readPixel(30,30),HEX);  // <----- Try to read color
}

void loop() {
}
```

commit acdfeb0b (2020-12-23): Fix #873

Minor performance boost to ESP32 block write