Bodmer/TFT_eSPI #878: Lvgl bugs the display

Trackhe, 2020-12-22:
![Uploading 102924606...]

Bodmer, 2020-12-22:
Do the TFT_eSPI examples work with the display?

Trackhe, 2020-12-23:
Yes

Bodmer, 2020-12-23:
OK, that suggests the problem is in your sketch.

Trackhe, 2020-12-23:
i try the spi display without touch in code. i will see if this happend when i use the spi display with the exakt same code

Trackhe, 2020-12-23:
The Display with SPI works with the exact same sketch.

Bodmer, 2020-12-29:
Can you post a copy of the sketch so I can test it and track the problem down?

Trackhe, 2020-12-29:
I try to reproduce it with an example sketch. in the next days.

Bodmer, 2021-01-06:
I loaded lvgl and configured it. I can confirm that the two examples (ESP32_TFT_eSPI and ESP32_TFT_eSPI_Slider) work with a SPI display.

The ESP32_TFT_eSPI example runs on and ESP32 with 8 bit parallel interface but the ESP32_TFT_eSPI_Slider example will not. I tried disabling the touch functions in that non-working sketch but could not get it to run at all. The same hardware setup does run the TFT_eSPI examples OK. So it appears there may be a conflict between the libraries for some reason in 8 bit parallel mode.  I do not see this as a TFT_eSPI library issue and thus will not be investigating further. In the case where someone identifes the need for a simple change then this will be examined for incorporation.

closed 2021-01-06 by Bodmer

Trackhe, 2021-01-06:
Thank you for your work.