Bodmer/TFT_eSPI #884: ESP32 with ILI9341 and XPT2046 - Two SPI busses?

trailsurfer604, 2020-12-24:
Hi,

For the life of me, I cannot get my 2.4" touch [display ](https://www.amazon.com/gp/product/B07WNLNRDN/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)to work with ESP32 WROOM32 and TFT_eSPI library.

The display has an ILI9341 controller and XPT2046 mounted on the PCB.  What seems odd (at least to me), is that the XPT2046 pins are connected to a separate set of SPI pins (see photos).

The screen works fine, but I cannot get any functionality out of the touch controller.  I am currently testing with the provided I am currently testing with the provided Test_Touch_Controller.ino, but all I get is 0,0,0.  I previously tried it with XPT2046_Touchscreen.h and touched() was active all the time even though nothing was touching the screen.

I have not modified the Test_Touch_Controller.ino. My custom user_setup (with font stuff omitted) and read-user-setup output is below, and I've attached photos of the display board.

Could it be that this board has XPT2046 on a separate SPI bus and, if so, how do I configure TFT-eSPI to run with this config?

Thanks in advance

**user_setup**
```
#define ILI9341_DRIVER
#define TFT_BL   16            // LED back-light control pin
#define TFT_BACKLIGHT_ON HIGH  // Level to turn ON back-light (HIGH or LOW)

// For ESP32 Dev board (only tested with ILI9341 display)
// The hardware SPI can be mapped to any pins

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS    5  // Chip select control pin
#define TFT_DC    0  // Data Command control pin
#define TFT_RST   2  // Reset pin (could connect to RST pin)
//#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST

#define TOUCH_CS 15     // Chip select pin (T_CS) of touch screen

#define SPI_FREQUENCY  27000000
// #define SPI_FREQUENCY  40000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000

// The ESP32 has 2 free SPI ports i.e. VSPI and HSPI, the VSPI is the default.
// If the VSPI port is in use and pins are not accessible (e.g. TTGO T-Beam)
// then uncomment the following line:
//#define USE_HSPI_PORT
![display back 2](https://user-images.githubusercontent.com/75198746/103048136-6e8fad80-4542-11eb-9e05-c1c5e880d30a.jpg)
![display back](https://user-images.githubusercontent.com/75198746/103048137-6f284400-4542-11eb-98fa-2869e51f96fd.jpg)
![display front](https://user-images.githubusercontent.com/75198746/103048139-6fc0da80-4542-11eb-82f8-730e70b8fb65.jpg)
```

**read-user-setup**
```
23:14:35.568 -> TFT_eSPI ver = 2.3.4
23:14:35.568 -> Processor    = ESP32
23:14:35.568 -> Frequency    = 240MHz
23:14:35.568 -> Transactions = Yes
23:14:35.568 -> Interface    = SPI
23:14:35.568 -> Display driver = 9341
23:14:35.568 -> Display width  = 240
23:14:35.568 -> Display height = 320
23:14:35.568 -> 
23:14:35.568 -> MOSI    = GPIO 23
23:14:35.568 -> MISO    = GPIO 19
23:14:35.568 -> SCK     = GPIO 18
23:14:35.568 -> TFT_CS   = GPIO 15
23:14:35.568 -> TFT_DC   = GPIO 0
23:14:35.568 -> TFT_RST  = GPIO 2
23:14:35.568 -> 
23:14:35.568 -> Font GLCD   loaded
23:14:35.568 -> Font 2      loaded
23:14:35.568 -> Font 4      loaded
23:14:35.568 -> Font 6      loaded
23:14:35.568 -> Font 7      loaded
23:14:35.601 -> Font 8      loaded
23:14:35.601 -> Smooth font enabled
23:14:35.601 -> 
23:14:35.601 -> Display SPI frequency = 27.00

```
[image]
[image]
[image]

elgerg, 2020-12-24:
You need to wire up the other 3 pins on the touch interface.

trailsurfer604, 2020-12-24:
> 
> 
> You need to wire up the other 3 (minus IRQ) pins on the touch interface.

how do you set pins for these connections in user-setup?

elgerg, 2020-12-24:
If memory serves you dont need to do anything. Just wire in the clock and data lines.

trailsurfer604, 2020-12-24:
OK, the answer was so obvious I missed it.

The 3 SPI lines need to be wired to the same bus.  Literally.  Using a breadboard as a splitter so that CLK (GPIO 18) goes to SCK and T_CLK on the display board, etc.

Now next issue is that I seem to be getting very odd readings at idle (i.e. nothing touching the screen.  For example:

`21:50:36.625 -> x: 338     y: 3384     z: 6 
21:50:36.866 -> x: 0     y: 3763     z: 16 
21:50:37.104 -> x: 0     y: 2932     z: 12 
21:50:37.374 -> x: 0     y: 3991     z: 27 
21:50:37.610 -> x: 389     y: 3393     z: 28 
21:50:37.848 -> x: 626     y: 3150     z: 21 
21:50:38.116 -> x: 64     y: 3535     z: 32 
21:50:38.352 -> x: 0     y: 3983     z: 27 
21:50:38.623 -> x: 344     y: 3383     z: 24 
21:50:38.861 -> x: 344     y: 3382     z: 24 
21:50:39.097 -> x: 0     y: 2916     z: 12 
21:50:39.368 -> x: 0     y: 3331     z: 25 
21:50:39.604 -> x: 0     y: 3711     z: 30 
21:50:39.872 -> x: 0     y: 3832     z: 23 `

elgerg, 2020-12-24:
I have noticed that sometimes the values i get are quite strange. I use the following to filter some of the noise:

bool my_touchpad_read(lv_indev_drv_t * indev_driver, lv_indev_data_t * data)
{
  uint16_t touchX, touchY;

  bool touched = tft.getTouch(&touchX, &touchY, 600);

  if (!touched)
  {
    return false;
  }

  if (touchX > screenWidth || touchY > screenHeight)
  {
    #if enableDebug
      Serial.println("Y or y outside of expected parameters..");
      Serial.print("y:");
      Serial.print(touchX);
      Serial.print(" x:");
      Serial.print(touchY);
    #endif
  }
  else
  {

    data->state = touched ? LV_INDEV_STATE_PR : LV_INDEV_STATE_REL;

    /*Set the coordinates (if released use the last pressed coordinates)*/
    data->point.x = touchX;
    data->point.y = touchY;
  }

  return false; 
}

I suspect that you are getting some strange values as you are using the breadboard as a splitter. Its probably introducing noise that is getting picked up as touch.

If i were you i would try to link the busses directly (ie solder a wire between them).

Hope that helps

trailsurfer604, 2020-12-26:
Thank you!  

It's working for now, and I will definitely solder them when I am out of prototyping stage.

For now, I have the screen connected to the ESP32 via a cat5e cable, plus a separate ground line (ignore the little switch)
[image]

closed 2020-12-26 by trailsurfer604