Bodmer/TFT_eSPI #891: readRect not working on TTGO T-Display (ST7789)

lesept777, 2020-12-28:
When I want to save a portion of the display in a buffer using readRect, the buffer is filled with 0s.

```
TFT_eSPI tft = TFT_eSPI(135, 240); // Invoke custom library
TFT_eSprite img = TFT_eSprite(&tft); // Sprite object

uint16_t bg[400];

void drawStar(int x, int y)
{
  img.setColorDepth(8);
  img.createSprite(20, 20);
  img.fillSprite(TFT_TRANSPARENT);
  img.drawLine(0, 0, 20, 20, TFT_RED);
  img.drawLine(0, 20, 20, 0, TFT_RED);
  img.pushSprite(x, y, TFT_TRANSPARENT);
  // Delete it to free memory
  img.deleteSprite();
}

void setup() {
  Serial.begin(115200);
  tft.begin();
  tft.setRotation(0); 
  tft.fillScreen(TFT_BLUE);
  int x = 60;
  int y = 10;
  // Move 1 sprite containing a "transparent" colour
  for (int i = 0; i < 100; i++)
  {
    tft.readRect(x, y, 20, 20, bg);
    if (i == 0) {
      for (int j = 0; j < 20; j++)
        for (int k = 0; k < 20; k++)
          Serial.printf ("%d %d : %d\n", j, k, bg[j * 20 + k]);
    }
    drawStar(x, y);
    delay(100);
    tft.pushRect(x, y, 20, 20, bg);
    ++y;
  }
}

void loop() {
}
```

All I see is a black square with a red cross moving down, and the buffer is filled with 0s
```
0 0 : 0
0 1 : 0
0 2 : 0
0 3 : 0
0 4 : 0
0 5 : 0
0 6 : 0
...
19 17 : 0
19 18 : 0
19 19 : 0
```

Bodmer, 2020-12-28:
The display has a bidirectional SPI data line, make sure you have this line in your setup file:
```
#define TFT_SDA_READ   // Display has a bidirectional SDA pin
```

Bodmer, 2020-12-28:
You will need to download the latest 2.3.54 version as well. I have released this version now so it will be available via the IDE library download option soon. You can otherwise download the latest direct from Github as a zip file.

lesept777, 2020-12-28:
Thanks, it's working!

closed 2020-12-28 by lesept777