Bodmer/TFT_eSPI #892: Setup11_Rpi do not work

Nulista, 2020-12-29:
Hallo Bodmer, 
Great library, Congrats!!

My project works fine with VERSION "2.3.2" and a 3.5" RPi display.
But when I install version 3.2.54 it doesn't work anymore .... only white screen.

For your information, if I omit lines 118 to 121 in User_Setup_Select.h, (just to find the problem..)
following lines:
----------------------
// Invoke 18 bit colour for selected displays
#if (defined (ILI9481_DRIVER) || defined (ILI9486_DRIVER) || defined (ILI9488_DRIVER)) && !defined (TFT_PARALLEL_8_BIT) && !defined (ESP32_PARALLEL)
  #define SPI_18BIT_DRIVER
#endif
--------------------
 then I get an incomplete black/white image.

Do you now wat i do wrong?
thanks

Bodmer, 2020-12-29:
Which processor are you using?

Nulista, 2020-12-29:
Esp32

Verstuurd vanaf mijn iPhone

Bodmer, 2020-12-29:
I have updated the master and this should be OK now.

closed 2020-12-29 by Bodmer