Bodmer/TFT_eSPI #893: V2.3.54 and  V2.3.55 not compiling for Parallel displays

randyjanzen, 2020-12-30:
Under v2.3.4 I had my ESP32 & ST7796 display working with a parallel interface. I updated the library and now it won't build. I reverted to 2.3.4 and it works. I tried with ILI9341 and  ST7796 user setups as well as my own.

To replicate my issue without introducing my own errors, I followed these steps:

1. compile & download TFT_Ellipse with v 2.3.4 using "My_User_Setup.h" based on User Setup 14 - Compiles and works on display.
2. delete library & reinstall version 2.3.54 or 2.3.55
2. select TFT_Ellipse example for 480x320 and compile with default user setup(SPI) - compiles without error
3. select User setup 14 in user User_Setup_Select.h - compile fails with errors:

_Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "Adafruit ESP32 Feather, 80MHz, 921600, None, Default"

In file included from C:\Users\randy\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.h:52:0,

                 from C:\Users\randy\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:17:

C:\Users\randy\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::begin_tft_write()':

C:\Users\randy\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.h:20:31: error: '_spi_user' was not declared in this scope

   #define SET_BUS_WRITE_MODE *_spi_user = SPI_USR_MOSI

                               ^
C:\Users\randy\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:64:3: note: in expansion of macro 'SET_BUS_WRITE_MODE'

   SET_BUS_WRITE_MODE;

   ^
C:\Users\randy\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::end_tft_write()':

C:\Users\randy\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.h:21:31: error: '_spi_user' was not declared in this scope

   #define SET_BUS_READ_MODE  *_spi_user = SPI_USR_MOSI | SPI_USR_MISO | SPI_DOUTDIN

                               ^
C:\Users\randy\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:84:29: note: in expansion of macro 'SET_BUS_READ_MODE'

   if(!inTransaction) {CS_H; SET_BUS_READ_MODE;}

                             ^
C:\Users\randy\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::begin_tft_read()':

C:\Users\randy\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.h:21:31: error: '_spi_user' was not declared in this scope

   #define SET_BUS_READ_MODE  *_spi_user = SPI_USR_MOSI | SPI_USR_MISO | SPI_DOUTDIN

                               ^
C:\Users\randy\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:107:3: note: in expansion of macro 'SET_BUS_READ_MODE'

   SET_BUS_READ_MODE;

   ^
C:\Users\randy\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::end_tft_read()':

C:\Users\randy\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.h:20:31: error: '_spi_user' was not declared in this scope

   #define SET_BUS_WRITE_MODE *_spi_user = SPI_USR_MOSI

                               ^
C:\Users\randy\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:340:3: note: in expansion of macro 'SET_BUS_WRITE_MODE'

   SET_BUS_WRITE_MODE;

   ^
exit status 1

Error compiling for board Adafruit ESP32 Feather._

Bodmer, 2020-12-31:
Thanks for raising this issue. I have been testing some changes to the SPI interface code and accidentally included these draft changes. I will have a look at the root cause and fix it.

Bodmer, 2020-12-31:
I have updated the master copy to correct this.

randyjanzen, 2020-12-31:
Unfortunately, it seems the update didn't work. I can run my display under 2.3.4, but not 2.3.56.

Bodmer, 2020-12-31:
Thanks for raising this and your patience. I will update but is you include this line in you setup file it will work OK:
```
#define TFT_PARALLEL_8_BIT
```
This is because this is now the favoured way of declaring a parallel display. I will update so that #define ESP32_PARALLEL will also work as that is included in some of the setup files.

closed 2020-12-31 by Bodmer

randyjanzen, 2020-12-31:
That is the define I used. I thought I saw somewhere in the processor code for ESP32, where you defined ESP32_PARALLEL if TFT_PARALLEL_8_BIT was defined. Also, I selected an unedited User Setup : #14 for my test compilation.

I will try the updated library at another time. 2.3.4 works for me.

commit 8e75cfa8 (2020-12-31): Fix #893
commit 64239999 (2020-12-31): Fix #893