Bodmer/TFT_eSPI #899: Stm SPIClass::SPIClass(int, int, int) missing

arnold-b, 2020-12-31:
Hello, 
i guess i use the wrong hardware library. which one is the right library ?

thanks

Bodmer, 2020-12-31:
You need[ the "official" core here](https://github.com/stm32duino/Arduino_Core_STM32).

arnold-b, 2020-12-31:
Then I´m confused. Because I´m using this one.

Bodmer, 2020-12-31:
Are you using the Arduino IDE?
Did you load the core via the Arduino IDE Boards Manager?
Do you have other STM32 board pacakges loaded that could cause a conflict?
I am using the STM32 core board package 1.9.0 and it compiles fine...
Post the relevant part of the compiler error output for review.

Bodmer, 2020-12-31:
Have you included #define STM32 in your setup file?
Example [here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup32_ILI9341_STM32F103.h).

Bodmer, 2021-01-06:
I have had a further look. The STM32 c[onstructor for the SPIClass is here.](https://github.com/stm32duino/Arduino_Core_STM32/blob/master/libraries/SPI/src/SPI.cpp#L28-L52).  Not that the comments indicate the requirements for the pin format and that the CS pin is optional.

The [TFT_eSPI calls the constructor here](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_STM32.c#L16). Since the CS pin is optional this call should work if the pins TFT_MOSI etc are given the right pin format.  The pin allocation example I have used is in this form and works fine:
```
#define TFT_MOSI PA7
#define TFT_MISO PA6
#define TFT_SCLK PA5
```
Since this all appears compliant with the constructor and works fine on my setup I suggest you check the correct setup file is being used by the compiler. To do this run the "Read_User_Setup" Test and diagnostics example. This only outputs to the serial port. My output when using the Setup29_ILI9341_STM32.h setup file is:
```
TFT_eSPI ver = 2.3.56
Processor    = STM32
Transactions = Yes
Interface    = SPI
Display driver = 9341
Display width  = 240
Display height = 320

MOSI    = GPIO 9999
MISO    = GPIO 100100
SCK     = GPIO 101101

>>>>> Note: STM32 pin references above D15 may not reflect board markings <<<<<
TFT_CS   = D5
TFT_DC   = D6
TFT_RST  = D7
TOUCH_CS = D4

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 55.00
Touch SPI frequency   = 2.50
``
The MOSI, MISO and SCK pins quoted look odd but this is a fault in the diagnostic sketch (fails to get the right pin reference and prints pin number twice. So pin reference number 99, 100 and 101 are actually being reported, this is for an STM32F767ZI Nucleo board, In practice the pin definitions are:

#define TFT_MOSI PA7
#define TFT_MISO PA6
#define TFT_SCLK PA5

So, I have found a bug in the diagnostics sketch for the STM32, this is not the cause of your issue though. I suggest you run the diagnostic sketch anyway and report the results.

arnold-b, 2021-01-06:
Thank you, i will try and test it

Bodmer, 2021-01-12:
Closing - assumed resolved.

closed 2021-01-12 by Bodmer