Bodmer/TFT_eSPI #905: [feature request] smooth/antialiased/alphablended  diagonal lines & rotations

DavyLandman, 2020-12-31:
I would really appreciate if the smoothing/alphablending support could be extended to more primitives. 

I'm thinking of a drawSmoothLine, but also for triangles and other primitives. I'm sure users can implement these by hand, but it would be nice to have this baked in and optimized like the rest.

And one where doing it manually takes quite some pain: pushRotated on sprites. I love this abstraction, but redoing that just to add alphablending is quite outside my comfort level with the math involved.

Bodmer, 2021-01-02:
I have some anti aliased drawing functions but they are not optimised for speed and thus are more for making static elements look pretty!

Do you have an example sketch from your project that needs the anti-aliasing primitives that I can experiment with. I don't want to add features that are not useful...

DavyLandman, 2021-01-02:
Rotating dials on a clock?

```cpp
#include <Arduino.h>
#include <TFT_eSPI.h> 
#include <SPI.h>

#include "NotoSansBold15.h"

TFT_eSPI lcd = TFT_eSPI();  // Invoke library, pins defined in User_Setup.h
TFT_eSprite face = TFT_eSprite(&lcd);
TFT_eSprite hourNeedle = TFT_eSprite(&lcd);
TFT_eSprite minuteNeedle = TFT_eSprite(&lcd);

constexpr uint32_t CLOCK_RADIUS = 128 / 2;
constexpr uint16_t CLOCK_COLOR_FACE = TFT_RED;
constexpr uint16_t CLOCK_COLOR_BG = TFT_WHITE;

constexpr uint32_t HOUR_ANGLE = 360 / 12;
constexpr uint32_t MINUTE_ANGLE = 360 / 60;

static double toRad(uint32_t angle) {
    return DEG_TO_RAD * angle;
}

static void renderFace() {
  face.createSprite(CLOCK_RADIUS * 2, CLOCK_RADIUS * 2);
  face.fillCircle(CLOCK_RADIUS, CLOCK_RADIUS, CLOCK_RADIUS, CLOCK_COLOR_BG);
  face.fillCircle(CLOCK_RADIUS, CLOCK_RADIUS, 3, CLOCK_COLOR_FACE);
  face.setTextDatum(MC_DATUM);
  face.loadFont(NotoSansBold15);
  face.setTextColor(CLOCK_COLOR_FACE); 
  constexpr uint32_t dialOffset = CLOCK_RADIUS - 10;
  for (uint32_t h = 0; h < 12; h++) {
      double x = CLOCK_RADIUS + (dialOffset * cos(toRad(h * HOUR_ANGLE))); 
      double y = CLOCK_RADIUS + (dialOffset * sin(toRad(h * HOUR_ANGLE)));

      uint32_t actualHour = (h + 3) % 12;
      face.drawNumber(actualHour == 0 ? 12 : actualHour, round(x), round(y));
  }
  face.unloadFont();
}

static void renderNeedles() {
  hourNeedle.createSprite(CLOCK_RADIUS / 2, 3);
  hourNeedle.fillSprite(CLOCK_COLOR_FACE);
  hourNeedle.fillRect(0,0, 3, 3, TFT_TRANSPARENT);
  hourNeedle.setPivot(0, 1);
  minuteNeedle.createSprite(CLOCK_RADIUS - 20, 3);
  minuteNeedle.fillSprite(CLOCK_COLOR_FACE);
  minuteNeedle.fillRect(0,0, 3, 3, TFT_TRANSPARENT);
  minuteNeedle.setPivot(0, 1);
}

void setup() {
  Serial.begin(74880); // native to debug output of bootloader
  lcd.init();
  lcd.setRotation(1);
  lcd.fillScreen(TFT_BLACK);
  renderFace();
  renderNeedles();
  lcd.setPivot(CLOCK_RADIUS, CLOCK_RADIUS);
}

static uint16_t hours = 0;
static uint16_t minutes = 0;
void loop() {
    face.pushSprite(0,0, TFT_TRANSPARENT);
    hourNeedle.pushRotated(hours, TFT_TRANSPARENT);
    minuteNeedle.pushRotated(minutes, TFT_TRANSPARENT);
    hours = hours >= 360 ? HOUR_ANGLE / 10 : hours + (HOUR_ANGLE / 10);
    minutes = minutes >= 360 ? MINUTE_ANGLE : minutes + MINUTE_ANGLE;
    delay(500);
}
```

DavyLandman, 2021-01-02:
I would think that having a subset of slower functions would be fine, as long as user can choose for them?

Maybe it's a bit outside of the scope of this library. I just got spoiled by the smooth font rendering that now I see edges everywhere 🙄

Bodmer, 2021-01-03:
Thanks I will have a look. I do have a function to draw wide anti-aliased lines with rounded ends so I think that would be good for the clock hands.  I have also implemented the WU algorithm for circles and lines. So the face circle could also be smoothed.

These functions are not in a library but I can patch then into your sketch for you to try.

DavyLandman, 2021-01-03:
To be honest, I simplified the sketch a bit, I use a custom picture as a
background (that my son designed).

Especially the rotating dials (which I did using sprites instead of
manually implementing WU) are the biggest ugliness.

but I'm happy to try out what can be done on this sketch example.

Bodmer, 2021-01-04:
Yes, I understand the jagged line problem. At close range this is quite visible but most of my displays are viewed from a distance of say 0.5m, so the steps are barely visible, especially on the higher resolution displays.

The effectiveness of anit-aliasing is also quite viewing angle and gamma setting dependant, most displays are best viewed in portrait mode.

One option is the dial is not anti-aliased until it slows down, then persistence of vision (and display refresh rate aliasing) means the jagged edges are not noticeable when the image is moving, then the peak rotation speed can be faster. Rotating dials are rather retro, based on old mechanical limits and designs but are fun to create in software. At the moment I am working on an anti-aliased version of [this dial style](https://content.instructables.com/ORIG/FDE/1SSL/I7CCD8DR/FDE1SSLI7CCD8DR.jpg?auto=webp&frame=1&fit=bounds&md=e1b223b0bf8fceb53bd727f09b5de782) which I prefer.

Bodmer, 2021-01-04:
I have put together an example based on your sketch using some AA functions, it works quite well. I have changed a few things, in particular the hand coordinates are now floating point values to get a nice smooth movement between the discrete pixel points.

The drawWideLineAA function will be built into TFT_eSPI eventually (maybe as a drawLine variant) with additional features. The ESP32 has a FPU but it is not very fast compared to other processors, however it is adequate for the slow moving clock hands.   An ESP8266 will probably be 10x slower to render the AA line as that processor has no FPU, but it woud still be OK for slow moving pointers in a sprite.
[Clock_905.zip](https://github.com/Bodmer/TFT_eSPI/files/5767274/Clock_905.zip)

DavyLandman, 2021-01-05:
wow, looks smooth :) thanks for all the work. I think the `drawWideLineAA` would be really nice to get inside TFT_eSPI indeed :)

closed 2021-01-05 by Bodmer

audreylim, 2022-02-21:
Was this ever integrated into TFT_eSPI or are there plans to do so?

Bodmer, 2022-02-21:
New functions have been added recently, only in the Github master at the moment. Emamples here:
https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Smooth%20Graphics

audreylim, 2022-02-21:
It works and looks great. Thank you!

DavyLandman, 2022-05-06:
Thanks for integrating it into master, it looks like versions from [2.4.42](https://github.com/Bodmer/TFT_eSPI/releases/tag/2.4.42) and higher should have this code.