Bodmer/TFT_eSPI #906: Can't get Wavesahre 4inch RPi LCD (C) to work with ESP32

MasT3r-A, 2021-01-01:
Hey Bodmer,

first of all - thanks for your TFT_eSPI library. Great Work.

I can't get this display

[https://www.waveshare.com/wiki/4inch_RPi_LCD_(C)](url)

to work with your library. I only get a white display. At one specific viewing angle i slightly see that there is something.

I thought this display is also a "good" display, except it is the 4" version 

Like you wrote in the info.

`The "good" RPi displays are the MHS-4.0 inch Display-B type ST7796 and Waveshare 3.5 inch ILI9486 Type C displays are supported and provides good performance. These have a dedicated controller and can be clocked at up to 80MHz with the ESP32 (55MHz with STM32 and 40MHz with ESP8266).`

Any known issues? I tried all combinations in the User_Setup.h. But nothing works.

My settings are:

MOSI             GPIO 23
MISO             GPIO 19
SCK                GPIO 18
TFT_CS           GPIO 15
TFT_DC          GPIO 2
TFT_RST         GPIO 4
TOUCH_CS     GPIO 22

I tried different SPI frequencies

Any suggestions? Or is that display just not supported?

Thanks in advance.

Greetz
MasT3r-A

MasT3r-A, 2021-01-01:
OK i changed the driver to 9488 (#define ILI9488_DRIVER ) and it's working now. 

But i also had to enable #define RPI_DISPLAY_TYPE (without it's not working)

Now i get 

`Display SPI frequency = 27.00`

Is this the max possible frequency?

Bodmer, 2021-01-02:
It should work at 80MHz with the ESP32.
Try also 40MHz if that does not work.

MasT3r-A, 2021-01-02:
Hey Bodmer,

thanks for your answer.

I overlooked that there is also a setting for the frequency in the User_Setups folder

When i changed to Setup21_ILI9488.h i forgot to change the frequency in that file. My bad :(

Ok now everything is working fine. 

Happy New Year to all of you!

Greetz

closed 2021-01-02 by MasT3r-A