Bodmer/TFT_eSPI #907: BMP image display

xtrinch, 2021-01-02:
https://github.com/Bodmer/TFT_eSPI/blob/64239999e4a703c8ec2e1da4a92f1e8b4a38f4df/examples/Generic/TFT_SPIFFS_BMP/BMP_functions.ino#L3

I've tried like 10 online converters into BMP format, but the format does not seem to be recognized.

It fails at this line: https://github.com/Bodmer/TFT_eSPI/blob/64239999e4a703c8ec2e1da4a92f1e8b4a38f4df/examples/Generic/TFT_SPIFFS_BMP/BMP_functions.ino#L33

These three values are consistent across different online converters and they are 1 32 3, not 1 24 0, as the code suggests. Can you give any pointers how to create an image that could be parsed by the snippet?

xtrinch, 2021-01-02:
The sooner I opened the issue, the sooner I probably would have figured it out 😆 

So I used image magick with the following settings to get the proper color bit depth and compression:
` magick convert -depth 24 -compress none image.bmp new-image.bmp`

Bodmer, 2021-01-03:
Thanks for the information. I have just used MS Paint in the past and the default setting for BMP generation works fine. I think IrfaView defaults were also OK.

It is useful to know what settings are required for image magick, so I will post to the discussions area.

closed 2021-01-03 by Bodmer