Bodmer/TFT_eSPI #910: DMA examples do not work with TTGO-T-Camera Plus

Kabron287, 2021-01-03:
All works in non DMA mode.
With DMA enabled, I got the following diagnostic:

> E (1979) spi_master: spi_bus_add_device(455): When GPIO matrix is used in full-duplex mode at frequency > 26.7MHz, device cannot read correct data.
Please note the SPI can only work at divisors of 80MHz, and the driver always tries to find the closest frequency to your configuration.
Specify ``SPI_DEVICE_NO_DUMMY`` to ignore this checking. Then you can output data at higher speed, or read data at your own risk.
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x4008f3a0
file: "D:\Arduino\libraries\TFT_eSPI-2.2.6\Processors/TFT_eSPI_ESP32.c" line 694
func: bool TFT_eSPI::initDMA()
expression: ret
Backtrace: 0x4008eec0:0x3ffb1ed0 0x4008f3a3:0x3ffb1ef0 0x400d3777:0x3ffb1f10 0x400d10cf:0x3ffb1f80 0x400d62c3:0x3ffb1fb0 0x4008bb75:0x3ffb1fd0

Bodmer, 2021-01-03:
The warning is happening because the TTGO-T-Camera Plus does not use the "normal" SPI pins, so the signals have to be routed via the I/O matrix which decreases the guaranteed speed of transfers. This warning is produced by the core board package not the TFT_eSPI library.

You have 2 options:
1. Reduce the SPI speed to 27MHz
2. Usea later TFT_eSPI version (current is 2.3.58) where the [warning is disabled here](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_ESP32.c#L714). Note however that the transfers may or may not be error free at frequencies above 27MHz.

Kabron287, 2021-01-04:
Thanks a lot it works!

closed 2021-01-04 by Kabron287