Bodmer/TFT_eSPI #912: How To different displays in use

middelbh, 2021-01-04:
I am using different displays in combination with ESP32 and Arduino. What possibilities do I have to use them without having to adapt the user setup in the library? Is it possible to set the correct display-data in the sketch?

Bodmer, 2021-01-04:
There is no option to do this at the moment with the Arduino IDE. With platform IO you can have a project specific "platformio.ini" file.

There are few boards that have a built in display, some of these could be auto-detected and this is a feature that may be added in future.  Most users seem to use bespoke setups though, as this is more cost effective compared to integrated solution and llarger displays are readily available.

To change the hardfware setup only needs a // to be added and a // to be removed in the User_Setup_Select.h file (if you create new or use the existing setup files). Typically this edit action takes only a few moments and compared to the other activities during development this time and effort is actually tiny. I mark a number on each hardware module which tells me which setup to use. The key advantage of this approach is that you can then run the provided examples without modification, so this can help with experimenting with the different functions, it also reduces option clutter in examples.  One library I tried to use recently had over 120 lines of configuration options at the start of the sketch which is crazy in my view.

As an input to future features, what are the differences between you setups (pins, display driver, processor, SPI port, parallel etc) ?

middelbh, 2021-01-04:
Hi, thx for your reply!
As I am working with several different displays and different ESP32 modules it would be extremely nice to have the setup for the specific display/ ESP combination in the sketch so I do not have to change the basic setup file everytime I use another combination..
Ideally the usersetup file would be located in the sketch directory, and only that file would have to be adapted for the display/ESP combination.

 If that is difficult to arrange,
Would it be possible to use
#include “TFT_eSPI.h” in stead of  #include <TFT_eSPI.h> and then store the entire library in the sketch directory, as in the first include (with quotes) the includefile will be searched in this directory in stead of in the standard library...

Grtz Hans

Bodmer, 2021-01-05:
I will add to my long TO DO list... thanks for you input and comments.

closed 2021-01-05 by Bodmer