Bodmer/TFT_eSPI #913: DMA examples do not work with blackpill (stm32f411ce)

MeKeCNC, 2021-01-04:
All works in non DMA mode.
Lib version: 2.3.58

I tried two different TFTs with blackpill, ili9341 and ili9486. generic examples work fine no problem. But when I want to use DMA, it gives the following error while compiling. As suggestion, is compatibility problem with balckpill 32f411ce and TFT_eSPI_STM32.h this library?

I used in main function this definitions to active DMA
#define USE_DMA_TO_TFT
#define STM32_DMA
#define STM32
**When compiling "boing poll" from sample applications for DAM use:**

Arduino: 1.8.13 (Windows 10), Board: "Generic STM32F4 series, BlackPill F411CE, STM32CubeProgrammer (SWD), Enabled (generic 'Serial'), CDC (generic 'Serial' supersede U(S)ART), Low/Full Speed, Smallest (-Os default), Newlib Nano (default)"
c:/users/powerpc/appdata/local/arduino15/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\PowerPC\AppData\Local\Temp\arduino_build_759726\sketch\boing_ball.ino.cpp.o: in function `setup':
boing_ball.ino.cpp:(.text.setup+0x34): **undefined reference to `TFT_eSPI::initDMA(bool)'**
c:/users/powerpc/appdata/local/arduino15/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\PowerPC\AppData\Local\Temp\arduino_build_759726\sketch\boing_ball.ino.cpp.o: in function `loop':
boing_ball.ino.cpp:(.text.loop+0x310): **undefined reference to `TFT_eSPI::pushPixelsDMA(unsigned short*, unsigned long)'**
collect2.exe: error: ld returned 1 exit status
Using library TFT_eSPI_2.3.58 at version 2.3.58 in folder: C:\Users\PowerPC\Documents\Arduino\libraries\TFT_eSPI_2.3.58 
Using library SPI at version 1.0 in folder: C:\Users\PowerPC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\libraries\SPI 
Using library SrcWrapper at version 1.0.1 in folder: C:\Users\PowerPC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\libraries\SrcWrapper 
exit status 1
Error compiling for board Generic STM32F4 series.

**When compiling "sprite Rotating Cube" from sample applications for DAM use:**

C:\Users\PowerPC\AppData\Local\Temp\arduino_build_840035\sketch\SpriteRotatingCube.ino.cpp.o: in function `setup':
SpriteRotatingCube.ino.cpp:(.text.setup+0x96): **undefined reference to `TFT_eSPI::initDMA(bool)'**
c:/users/powerpc/appdata/local/arduino15/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\PowerPC\AppData\Local\Temp\arduino_build_840035\sketch\SpriteRotatingCube.ino.cpp.o: in function `loop':
SpriteRotatingCube.ino.cpp:(.text.loop+0x100): **undefined reference to `TFT_eSPI::dmaBusy()'**
c:/users/powerpc/appdata/local/arduino15/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: SpriteRotatingCube.ino.cpp:(.text.loop+0x132): **undefined reference to `TFT_eSPI::pushImageDMA(long, long, long, long, unsigned short*, unsigned short*)'**
collect2.exe: error: ld returned 1 exit status
Using library TFT_eSPI_2.3.58 at version 2.3.58 in folder: C:\Users\PowerPC\Documents\Arduino\libraries\TFT_eSPI_2.3.58 
Using library SPI at version 1.0 in folder: C:\Users\PowerPC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\libraries\SPI 
Using library SrcWrapper at version 1.0.1 in folder: C:\Users\PowerPC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\libraries\SrcWrapper 
exit status 1
Error compiling for board Generic STM32F4 series.

Bodmer, 2021-01-04:
Check the following:
1. DMA only works with SPI displays
2. #define STM32 must be present in your setup file inside the library. Placing in a sketch will have no effect due to the way the Arduino environment works.

You can check the compiler is picking this up with the "Read_User_Setup" diagnostics sketch.

Bodmer, 2021-01-04:
#define STM32_DMA  must be present in your setup file inside the library too.

Bodmer, 2021-01-08:
I have setup and run an STM32F411 "Black Pill" board and it runs fine with DMA, so this may be a software setup issue you are having. Post back your findings.

Bodmer, 2021-01-12:
Assumed resolved.

closed 2021-01-12 by Bodmer