Bodmer/TFT_eSPI #919: ili9488 black color weird issue

horvathpeterb610, 2021-01-06:
Hi!
So first of all thanks for your work!

My problem is really weirdo.
So the black color flickering only when tft.fillscreen is black other colors no problem, see the first pics.
But if i switch the TFT_INVERSION_ON and all the color switch in the code, no problem?!?!
Its a pain to inverse all the bmp and colors to good.
Any idea what can it be the problem?
[Inversion off](https://imgur.com/QWoZzAt)
[Inversion on](https://imgur.com/aCl8Qba)

Thanks
Eszement

Bodmer, 2021-01-06:
Yes, that is very odd. This implies that the display behaves differently clocking in a lot of "zero" data values (when invert is off to get black) compared to a lot of "ones" data values when inversion is on. The patterns on the display suggests like the internal bias levels within the display are being changed, as these are programmable and perhaps are getting corrupted.

Here is a  few guesses at things to try:
1.  Signal integrity may be the issue, try adding extra GND wires between the processor (ESP32?) and the display. Also reduce the wire lengths (ESP32 for example outputs very high edge speeds which causes "ringing" on clock edges. Remove any wire stubs.
2. Remove any SD card in the display, this may be affecting signals to the display especially if the chip select to the SD card has not been set high.
3. Try reducing the SPI clock rate.
4. Try a different 5V/3.3V power supply supply to power the display (which are you using?). This test is in case the power supply demands may be different when clocking in 0's or 1's
5. Check the vendors advert and confirm the dislay really is an ILI9488 type (some vendors adverts  indicate you may get a random driver choice from their stock!).
6. Test another display, ideally from a different purchase batch.

These are things I would try in your situation but, as you say, this one is rather odd!

horvathpeterb610, 2021-01-06:
Hello!
Thanks for the quick reply!
So its an esp8266.
1. Extra gnd same result.
2. Not use sdcard slot.
3. Try but same result
4. Try it with labor bench supply same result 
5. How can i check the display driver? any readreg with esp?
6. So i bought 5 display 3 no problem with inversion off, the fourth start the flickering really slow day by day , and the fifth is flicker at the start, same cable lenght same soldering.

Bodmer, 2021-01-06:
Post a link to the seller web page so I can see the PCB and component details and compare to my display.

List all the connections you have used between the ESP8266 and display.

You could try the "Read_ID_bitbash" sketch to see the registers but that sketch is old and written for smaller displays and only reads certain registers. It might or might not work in your setup.

horvathpeterb610, 2021-01-06:
Heres the link: [Aliexpress](https://a.aliexpress.com/_qeGeLlOd)
I follow the main setup so:  
 Display SDO/MISO  to NodeMCU pin D6 (or leave disconnected if not reading TFT)
 Display LED            to NodeMCU pin 3.3V
 Display SCK            to NodeMCU pin D5
 Display SDI/MOSI  to NodeMCU pin D7
 Display DC (RS/AO)to NodeMCU pin D3
 Display RESET     to NodeMCU pin esp RST
 Display CS        to NodeMCU pin D8 
 Display GND       to NodeMCU pin GND (0V)
 Display VCC       to NodeMCU 3.3V

And the report:
00:33:30.016 -> Register 0x01: 0x00
00:33:30.110 -> Register 0x04: 0xFFFFFF
00:33:30.110 -> Register 0x09: 0xFFFFFFFF
00:33:30.110 -> Register 0x0A: 0xFF
00:33:30.110 -> Register 0x0B: 0xFF
00:33:30.110 -> Register 0x0C: 0xFF
00:33:30.110 -> Register 0x0D: 0xFF
00:33:30.110 -> Register 0x0E: 0xFF
00:33:30.110 -> Register 0x0F: 0xFF
00:33:30.110 -> Register 0x2E: 0xFFFFFF
00:33:30.110 -> Register 0x3F: 0xFFFFFFFF
00:33:30.110 -> Register 0xDA: 0xFF
00:33:30.110 -> Register 0xDB: 0xFF
00:33:30.156 -> Register 0xDC: 0xFF
00:33:30.156 -> ===========================
00:33:30.156 -> 
00:33:30.156 -> Looks like driver chip is: Unknown

horvathpeterb610, 2021-01-06:
Went the aliexpress comment section and one guy wrote the ST7796 driver.
I try it and now good! 
So thanks for the idea and help 😄

Bodmer, 2021-01-07:
Great, note that those displays are very good for performance but some have a hardware issue [as detailed here](https://github.com/Bodmer/TFT_eSPI/discussions/898).

closed 2021-01-07 by Bodmer