Bodmer/TFT_eSPI #923: SD not working after getTouch because SPI is in wrong mode

EmbeddedEric, 2021-01-07:
Arduino, Libraray-Version 2.3.56
Board: ESP32 with 2,8" TFT-Display with ILI9341, Touch and SD.

All three peripherals are Connected to the same SPI (SPI3) with different CS and are working.

I'm not sure if this really is a bug, but,
if i read touchpoints with getTouch() the SPI has an other configuration then before.
Clock (26,6Mhz -> 2,5Mhz), usr_miso (1 -> 0), usr_mosi (1 ->0), doutdin (1 -> 0), usr_command (0 -> 1) are adjusted.

With the new configuration, sd library will not work and you get "file not found" messages.

Maybe it is possible to restore spi settings after using touch ?

Workarounds:
- Write something to the TFT (your driver adjust spi configuration to now working parameters, also for SD)
- reconfig SPI3 to suiteable values like..

```
void ReconfigSPI()
{
    // Mode
    SPI3.user.val = 0;
    SPI3.user.usr_mosi = 1;
    SPI3.user.usr_miso = 1;
    SPI3.user.doutdin = 1;

    // Clock to 26,6Mhz
    SPI3.clock.clkcnt_n = 2;
    SPI3.clock.clkdiv_pre = 0;
}
```

Bodmer, 2021-01-08:
The TFT_eSPI library releases SPI bus at the end of a screen ot touch access, the SD library should reconfigure the SPI bus (clock rate mode etc for when access to the SD card is requested.

I do not use the SD card interface so it is possible that the library does not release the bus correctly. To check this I need to replicate your setup. Please provide all pin to pin connections and software environment details: Arduino IDE version, in particular which SD library you are using (there are several).

EmbeddedEric, 2021-01-08:
You're right in that, that sd library should reconfigure the spi, but sd library does not.
So i think it's more an feature request than a bug.

ARDUINO-IDE v 1.8.13
ESP32-Package: 1.0.4
Used SD-Card-Lib, original from ESP32-Package (#include <SD.h>) 

Touch-Part:
T_OUT -> GPIO 19
T_DIN -> GPIO 23
T_CS  -> GPIO 21
T_CLK -> GPIO 18

TFT-Part:
LED   -> 3V3
SCK   -> GPIO 18
SDI   -> GPIO 23
D/C   -> GPIO 2
RESET -> GPIO 4
CS    -> GPIO 15
GND   -> GND
VCC   -> 3V3

SD-Part:
CLK   -> GPIO 18
MISO  -> GPIO 19
MOSI  -> GPIO 23
CS    -> GPIO 5

User_setup.h
```
#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)
#define TOUCH_CS 21     // Chip select pin (T_CS) of touch screen
```

```
void setup() 
{
  Serial.begin(115200);

  // Set all chip selects high to avoid bus contention during initialisation of each peripheral
  digitalWrite(22, HIGH); // Touch controller chip select (if used)
  digitalWrite(15, HIGH); // TFT screen chip select
  digitalWrite( 5, HIGH); // SD card chips select, must use GPIO 5 (ESP32 SS)

  if (!SD.begin()) 
  {
    Serial.println("Card Mount Failed");
    return;
  }
  
  m_pTft->begin();
  m_pTft->setRotation(1);  // landscape
  // now load an Image from SD to an Array and decode it with TJpg_decoder :-)
  // File jpg = SD.open("/1.jpg", FILE_READ); -> works
}

void loop()
{
   // load something from SD works (in the first loop)
   // File jpg = SD.open("/1.jpg", FILE_READ); -> works
   
   uint16_t x, y;   
   if (m_pTft->getTouch(&x, &y))
   {
       // File jpg = SD.open("/1.jpg", FILE_READ); -> fails
       // After draw something to the display, SD.open works again.
       // or set SPI Mode like first post.
   }
}
```

[image]

Bodmer, 2021-01-08:
See #915.

I will update the library. Post back if this works or not.

commit 5f171eee (2021-01-08): Fix #923 and #915
closed 2021-01-08 by Bodmer