Bodmer/TFT_eSPI #927: ESP32 - ILI9486 4" display

pawelmielnik, 2021-01-09:
I have been tried to use your TFT_eSPI-Library-ILI9486 on
 SKU: 13587
Part Number: 4inch TFT Touch Shield
Brand: Waveshare
UPC: 614961951242
 4" display.
 Unfortunatelly it dosn't work , do you know what changes should i do to run it on 4" display, or do you know any other existig solution.
Pawel,

Bodmer, 2021-01-09:
It looks like it has a 16 bit serial to parallel circuit. Maybe you could use with the raspberry Pi board setup file "Setup11_RPi_touch_ILI9486.h" suitably modified to drive the pins you are using.

I cannot say whether the board will work with the TFT_eSPI library or not, but it is worth a try.

pawelmielnik, 2021-01-10:
Hi
Thank you for advice, I will try.
Display works with Waveshare library on Arduino.
What wanders me, is the connection to Arduino Mega. I'm using this same
connection  as in Arduino (3-13 Waveshare recommendation), which means the
SPI interface in Mega is not used ( pin 50-54 whatever),  but the display
works....
Pawel,

pawelmielnik, 2021-01-11:
Hi,
It didn't help at all, I keep trying to find a solution.
One more question: what kind of LCD_Touch min 3,5" SPI working with ESP32
do you recommend and have you tested?
I'm using a very complex solution where I need WiFi  and can't use the set
of ADC2 on ESP32. Usage of WiFi blocks ADC2.
I also tried  8357 D 4" which is ok but i have to use ADC2 pins for Touch
(Ym,Yp,Xm,Xp), which doesn't resolve my issue.
Pawel,

niedz., 10 sty 2021 o 11:05 Pawel Mielnik <pawel.mielnik@gmail.com>
napisał(a):

> Hi
> Thank you for advice, I will try.
> Display works with Waveshare library on Arduino.
> What wanders me, is the connection to Arduino Mega. I'm using this same
> connection  as in Arduino (3-13 Waveshare recommendation), which means the
> SPI interface in Mega is not used ( pin 50-54 whatever),  but the display
> works....
> Pawel,
>
>
>
>> It looks like it has a 16 bit serial to parallel circuit. Maybe you could
>> use with the raspberry Pi board setup file "Setup11_RPi_touch_ILI9486.h"
>> suitably modified to drive the pins you are using.
>>
>> I cannot say whether the board will work with the TFT_eSPI library or
>> not, but it is worth a try.
>>
>>
>
>
> Best Regards
> Pawel Mielnik

Bodmer, 2021-01-11:
Some info can be found in the ReadMe.

[ST7796 SPI based displays](http://www.lcdwiki.com/4.0inch_SPI_Module_ST7796) are  good but need a PCB modification if you wish to use with other devices on the SPI bus (including a touch screen controller), [see here](https://github.com/Bodmer/TFT_eSPI/discussions/898).

The "good" RPi displays are the [MHS-4.0 inch Display-B type ST7796](http://www.lcdwiki.com/MHS-4.0inch_Display-B) and [Waveshare 3.5 inch ILI9486 Type C ](https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(C))displays are supported and provides good performance. These have a dedicated controller and can be clocked at up to 80MHz with the ESP32 (55MHz with STM32 and 40MHz with ESP8266).

Bodmer, 2021-01-11:
If you need a consistent supplier for continued availability then I have found [BuyDisplay ](https://www.buydisplay.com/)are good and provide documentation.

closed 2021-01-11 by Bodmer