Bodmer/TFT_eSPI #935: Multiple ILI9341 displays at 40MHz (or more)

bitwerkdk, 2021-01-12:
We are trying to get the highest possible frame rate on two **ILI9341** displays, with a program running on an **ESP32**.

Control of the two screens is achieved by commenting out `TFT_CS` in `User_Setup.h`, and setting the displays we want to write to manually in the program.

This all works fine at **27MHz**, at which we can achieve a frame rate around 10 FPS.
But once we increase the `SPI_FREQUENCY` beyond **27MHz**, rendering only happens on one of the displays :(

**Anyone have an idea why this happens, or maybe some tips as to how we can improve our frame rate?**

Here's the code:
```
#include <Arduino.h>
#include <SPI.h>
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI();

// Use instead of TFT_CS, which is commented out in User_Setup.h
const unsigned int TFT_CS1 = 15;
const unsigned int TFT_CS2 = 5;

unsigned long startTime;
unsigned long counter;

void setup() {
  pinMode(TFT_CS1, OUTPUT);
  pinMode(TFT_CS2, OUTPUT);
  digitalWrite(TFT_CS1, LOW);
  digitalWrite(TFT_CS2, LOW);

  // Setup the LCD
  tft.init();
  tft.setRotation(3);
  tft.fillScreen(0xFFFF);

  // Disable screens
  digitalWrite(TFT_CS1, HIGH);
  digitalWrite(TFT_CS2, HIGH);

  startTime = millis();
}

void loop() {
  randomSeed(millis());
  counter++;
  
  // Write to display 1
  digitalWrite(TFT_CS1, LOW);
  digitalWrite(TFT_CS2, HIGH);

  tft.fillScreen(counter);
  tft.setTextColor(TFT_BLUE+TFT_GREEN,0xF800);
  tft.drawCentreString("FPS: ", 160, 210, 2);
  tft.setTextDatum(TC_DATUM);
  tft.drawNumber(counter * 1000 / (millis() - startTime), 160, 225, 2);
  
  // Write to display 2
  digitalWrite(TFT_CS1, HIGH);
  digitalWrite(TFT_CS2, LOW);

  tft.fillScreen(counter);
  tft.setTextColor(TFT_BLUE+TFT_GREEN,0xF800);
  tft.drawCentreString("FPS: ", 160, 210, 2);
  tft.setTextDatum(TC_DATUM);
  tft.drawNumber(counter * 1000 / (millis() - startTime), 160, 225, 2);
}
```

bitwerkdk, 2021-01-12:
Now I'm even more confused. 

If I swap the screens both works at 40Mhz running at 15 FPS. 
Swapping them back and forth provides consistent results. Could there be some sort of issue with one of the screens?

I've seen people talking about replacing the apparently slow digitalWrite with a direct memory change of the ports of the CS pins. I wonder if that could provide further improvements to the FPS.

Bodmer, 2021-01-12:
Two displays can be used like that.  I assume you are using an ESP32?

Keep the wires short (<100mm total length)  becuase you can get signal integrity problems otherwise due to the high signal edge speeds. Power each display direct from the same power source, do not "daisy chain" the supplies, e.g do not connect power to display one and then power from display one to display two. Add extra GND (0V lines) from each display to the power source to reduce "ground bounce" from affecting signal integrity.  Multiple devices on a bus can "load" the bus with parasitic capacitance and this can impact the maximum transfer speed, however with just two displays I would not expect this to be a problem.

16 fps for each display (30fps total update speed) is about as fast as you can get for a ILI9341 at 40MHz SPI clock rate. It is easy to calculate:

Total bits sent for a 240x320 screen fill is: 240 * 320 * 16 = 1228800 bits
Time to fill a frame at 40MHz  is: 1228800/4000000 = 0.03072 seconds = 30.72 milliseconds = 32.55 fps maximum

Bodmer, 2021-01-12:
I tried this sketch on one of my 240*320 displays at 40MHz SPI clock rate on ESP32:
```
#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();

unsigned long startTime;
unsigned long counter;

void setup() {
  Serial.begin(250000);
  tft.init();
  tft.setRotation(3);
  tft.fillScreen(0xFFFF);
  startTime = millis();
}

void loop() {
  counter++;
  tft.fillScreen(counter);
  if(counter%100 == 0) Serial.println( counter * 1000.0 / (millis() - startTime) );
}
```

Result was 32.07 fps, so that is the limit for the library when used with that processor and no DMA.

bitwerkdk, 2021-01-12:
Great, thank's a lot  👍

closed 2021-01-12 by Bodmer