Bodmer/TFT_eSPI #937: STM32F411CE with ILI9341

Magix35, 2021-01-14:
Hello,
I can't display on a ili9341 with an stm32f411ce "black pill"  I have test with an stm32F103cb "blue pill" and all works fine using : 
TFT_CS   PA1
TFT_DC   PA0
TFT_RST  PA2
TFT_MOSI PA7
TFT_MISO PA6
TFT_SCLK PA5

and with the same connections to the stm32f411ce I have nothing the screen still white. I don't understand because the pins seems correct.

Bodmer, 2021-01-14:
I have used Setup32 and it works fine on a black pill. What version of TFT_eSPI are you running? Does an uploaded sketch run (e.g. "TFT_graphicstest_one_lib" still run and report to the Serial Monitor window even if nothing is displayed on the TFT?

Bodmer, 2021-01-14:
I see you have swapped PA1 and PA0 pin functionss compared to Setup32, was this intentional?

Magix35, 2021-01-14:
> I see you have swapped PA1 and PA0 pin functionss compared to Setup32, was this intentional?

I have connect the LCD  following the pin mapping the that you have choise in one of your issue.
I run the sketch Mandelbrot and there is nothing that output from the serial monitor.
I use PlatformIO so I use the last version of the library

Bodmer, 2021-01-14:
It will take 3 or 4 seconds to run the Mandlebrot example and then it only prints out one number at 250000 baud. It sounds like a PlatformIO environment setup problem because serial output should work even without a display connected...

How are you putting the black pill board into programming mode?
Are you sure the program is uploading OK?
Create a simpler sketch that just outputs serial messages, and get that running.

Magix35, 2021-01-14:
I am testing the Setup32 that you talk. I will tell you if that work after.
I use Serial for programming.

Magix35, 2021-01-15:
Hello, sorry but it wasn't the library that was the problem but it was my configuration of platformIO.
Since I'm using the GenericSTM32F411CE I had needed to define HSE_VALUE to 25MHz.

So I have to add in platformIO ini:
`build_flags = 
    -DHSE_VALUE=25000000U`

With that all work fine.
Thanks for your help.

closed 2021-01-15 by Magix35

Bodmer, 2021-01-15:
Good to know, thanks for posting the solution.