Bodmer/TFT_eSPI #938: TFT_eSPI wrong colours, origin and rotation with Arduino 1.8.13 (OK on 1.8.9)

lee-xp, 2021-01-16:
I have a number of sketches which work well on an ESP32 TTGO T-Display 135x240 TFT board with Arduino 1.8.9 (on Windows 10), but with 1.8.13, which I have just installed, the colours are very different (they look like they are inverted - almost) for both text and BMPs, the rotation is 90 degrees out, and the origin is wrong. In User_Setup_Select.h I am using
    #include <User_Setups/Setup25_TTGO_T_Display.h>
as I did with 1.8.9, and I am compiling using the ESP32 Dev Module board - again, as with 1.8.9.

I thought that this might be a library issue, so I used a copy of the sketchbook libraries (where TFT_eSPI is located) from 1.8.9. But this had no effect.

One sketch which demonstrates the issue is TFT_Meter_5 in the 160 x 128 folder of the library examples (with M_SIZE set to 1).

[image]

I have a feeling I have missed something very obvious, but can't think what.

Bodmer, 2021-01-16:
The change Arduino IDE version should not have an effect on how the TFT_eSPI library behaves.

What version of the TFT_eSPI library are you using in each case?

Run the "Read_User_Setup" sketch and verify the compiler is picking up the right setup file. The sketch sends the diagnostic information to the serial monitor window, not the display.

lee-xp, 2021-01-17:
Excellent! All fixed. Read_User_Setup is a very useful diagnostic tool. My problem was that I had not commented out the default setup #include <User_Setup.h>. In my defence, I did temporarily replace the sketchbook library contents of 1.8.13 with those of 1.8.9, and for some reason that didn't fix the issue - which it obviously should have - so I am not sure what happened there.

Many thanks.

closed 2021-01-17 by lee-xp