Bodmer/TFT_eSPI #940: Doesn't work for ESP32 and certain pinout configurations

maxgerhardt, 2021-01-17:
See https://community.platformio.org/t/tft-with-wemos-r32/18532/11?u=maxgerhardt.

Basically, when using a Wemos D1 R32 board (as pictured in the README of this repo, too) and a certain [display](http://www.lcdwiki.com/3.5inch_Arduino_Display-UNO), the library can impossibly work. 

Consider the pinout / config

```
build_flags = 
  -DUSER_SETUP_LOADED=1
  -DESP32_PARALLEL=1
  -DTFT_PARALLEL_8_BIT=1
  -DILI9486_DRIVER=1
  -DTFT_WIDTH=320
  -DTFT_HEIGHT=480
  -DTFT_RST=4
  -DTFT_CS=15
  -DTFT_RS=33
  -DTFT_WR=32
  -DTFT_RD=39
  -DTFT_D0=12
  -DTFT_D1=13
  -DTFT_D2=26
  -DTFT_D3=25
  -DTFT_D4=17
  -DTFT_D5=16
  -DTFT_D6=27
  -DTFT_D7=14
```

Note that the WR pin is IO32 (if I read the mapping correctly).

During compilation there are many errors such as 

```
<command-line>:0:8: warning: left shift count >= width of type [-Wshift-count-overflow]
.pio\libdeps\esp32_2\TFT_eSPI\Processors/TFT_eSPI_ESP32.h:187:40: note: in expansion of macro 'TFT_WR'
     #define WR_H GPIO.out_w1ts = (1 << TFT_WR)
```

because in this case `TFT_WR` is >= 32, it should be writing to `GPIO.out1_w1ts` with a `(1 << (TFT_WR - 32)` bitmask.

The current logic  around writing 16-bit data to the LCD with code like 

```
    #define tft_Write_16(C) GPIO.out_w1tc = clr_mask; GPIO.out_w1ts = set_mask((uint8_t) ((C) >> 8)); WR_H; \
                            GPIO.out_w1tc = clr_mask; GPIO.out_w1ts = set_mask((uint8_t) ((C) >> 0)); WR_H
..
  // Mask for the 8 data bits to set pin directions
  #define dir_mask ((1 << TFT_D0) | (1 << TFT_D1) | (1 << TFT_D2) | (1 << TFT_D3) | (1 << TFT_D4) | (1 << TFT_D5) | (1 << TFT_D6) | (1 << TFT_D7))

  // Data bits and the write line are cleared to 0 in one step
  #define clr_mask (dir_mask | (1 << TFT_WR))
```

doesn't work when the pins D0...D7 and WR lie in different "GPIO registers", so e.g. here WR=32 and all D0..D7 in the range <32. 

There needs to be some kind of logic here that writes the GPIO registers for the <32 pin number pins and then the >=32 pin number pins (if there exists), but the current code assumes that all of them are in the <32 range. And for exactly that LCD (if you want to sit it in the Arduino header without jumper-wiring everything to nicer pins) doesn't work.

The errors are of course gone once `TFT_WR` is set to <32.

maxgerhardt, 2021-01-17:
Okay I've noticed the comment

```cpp
////////////////////////////////////////////////////////////////////////////////////////
// Define the WR (TFT Write) pin drive code
////////////////////////////////////////////////////////////////////////////////////////
#if defined (TFT_WR)
  #if (TFT_WR >= 0)
    // TFT_WR, by design, must be in range 0-31 for single register parallel write
    #define WR_L GPIO.out_w1tc = (1 << TFT_WR)
    #define WR_H GPIO.out_w1ts = (1 << TFT_WR)
  #else
    #define WR_L
    #define WR_H
  #endif
#else
  #define WR_L
  #define WR_H
#endif
```

and I agree with that too

Bodmer, 2021-01-17:
If you cannot map the TFT_WR pin to be <32 then there is a simple change to let higher GPIO pins to be used. Post back if you need this.

maxgerhardt, 2021-01-17:
You're right maybe it does make sense in that special case to do a special code modification. 

IsN't the problem just with the `WR_L`, `WR_H` and `clr_mask` macros?

Bodmer, 2021-01-17:
I have pushed a change to the repo. Data GPIO pins must still be in the range 0-31
The graphicstest runs ~1.23x slower if the TFT_WR pin is allicated to a GPIO > 31, so bear this in mind if you need slightly better performance.

maxgerhardt, 2021-01-17:
Thanks for the changes! 

Now a really stupid thing is that I re-read the pinmapping a third time and figured out I was reading the analogue side of the pins backwards and `TFT_WR=4` is actually correct.. That is if I'm not stupid again and read it wrong now, which is not totally out of the question.

I've created a project for this display at https://github.com/maxgerhardt/pio-wemos-d1-r32-ili9486-display and gave it to the initially-asking user (who has the display) for testing.  

But even if it wasn't necessary for this, if I read the code correctly with the compile-time defines, for TFT_WR <=31 it should still use the same old fast code, right?

Bodmer, 2021-01-17:
OK, it was a change I had meant to do at some point.

Yes, TFT_WR <=31 will use the faster code.

Bodmer, 2021-01-18:
Note that you cannot use GPIO pin 39, it is input only! Note there is no recognition of "TFT_RS" you must use TFT_DC

Correct the pinout to be compatible with:
```
// ESP32 pins used for the parallel interface TFT
#define TFT_CS   33  // Chip select control pin
#define TFT_DC   15  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  32  // Reset pin

#define TFT_WR    4  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD    2

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits
#define TFT_D2   26
#define TFT_D3   25
#define TFT_D4   17
#define TFT_D5   16
#define TFT_D6   27
#define TFT_D7   14
```

closed 2021-01-19 by Bodmer