Bodmer/TFT_eSPI #941: BMP with indexed colors ?

sn00zerman, 2021-01-17:
Hello,

I really like the TFT_eSPI lib and also enjoy the TFT_eFEX :-) 
I'm currently building an opensource modelrailroad project with the help of this library. 
However, I have 1 issue left, to finish my current project ...
I need to download images from the internet, from a loco-image-filefault to SPIFFs. This works, but I'm unable to show the images.
They are in BMP-format, but with indexed colors. (Downloading works fine, when I retrieve the images from SPIFFs to my PC,
I can view them with Gimps and other image-viewers)

I tried to modify the code that shows the BMP files on screen (ILI9488 in my case), but I only get some gray imagebox,
with some random noise in it. (the dimensions in screen are however correct)

Can anyone help me out with this ? There is no way to retrieve the images in another format but BMP indexed colors, because I don't own the loco-image-filevault.
(I also can't find another Arduino library that takes care of indexed colors images, everybody seems only be interested in 24-bit or 16-bit images)

Please help ! 

best regards,
Kris
https://www.digitalplayground.be

Bodmer, 2021-01-17:
An interesting project. I have tinkered with indexed colours BMP files. The [code is here](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/ePaper/Floyd_Steinberg/Floyd_Steinberg_BMP.ino). In that example the image is a greyscale one but the pallette retrieval is the same process.

If you think you will not be able to adapt this code then upload an image example (or provide a link to one) and I will look at the format.

I think indexed BMP files are quite rare these days so I assume the images are quite old in origin.

sn00zerman, 2021-01-17:
Hello Bodmer,

I did have a look at the code, but I don't know what to do with the instructions containing an object called "frame"...
Included you may find 5 images:
https://www.digitalplayground.be/Bodmer/loco-00.bmp
https://www.digitalplayground.be/Bodmer/loco-01.bmp
https://www.digitalplayground.be/Bodmer/loco-02.bmp
https://www.digitalplayground.be/Bodmer/loco-03.bmp
https://www.digitalplayground.be/Bodmer/loco-04.bmp

Unfortunate it's not an old system. In the modelrailroad community, manufacturers still keep using "old formats" :-(
Even more strange, whenever a user uploads an image in RGB format, it gets "down-converted" to a 256-colors image !
(All images do have the same size: 190x40 pixels)

It would be nice to get a modified TFT_eFEX.h and TFT_eFEX.CPP, that includes 2 methods. (one for the 8-bit, and one for the 16/24-bit ... In our own software we do use RGB images)

Thank you for your help in advance. (wow, and talking of a quick response, and this on a sunday !!!)

Best regards,
Kris

Bodmer, 2021-01-17:
The attached sketch contains the function you need to finsh you project. I tested with one of your examples so it seems to work OK.
[BMP_941.zip](https://github.com/Bodmer/TFT_eSPI/files/5826630/BMP_941.zip)

sn00zerman, 2021-01-17:
Hi Bodmer,

Works like a charm now :-)
Thanks a million times ! This was the last issue to full-fill our project.
When we reach a milestone, I will inform you about an article of the project, and a potential youtube movie.
And of course, I will put your name and TFT_eSPI project in the credits :-)

thank you,
Kris

sn00zerman, 2021-01-17:
issue closed !

closed 2021-01-17 by sn00zerman

Bodmer, 2021-01-18:
Great, looking forward to seeing the movie.