Bodmer/TFT_eSPI #942: Bash error

bazfromleek, 2021-01-18:
I get following error when I try to compile an example (Using a MAC Big Sur).

bash: line 0: [: too many arguments
usage: cp [-R [-H | -L | -P]] [-fi | -n] [-apvXc] source_file target_file
       cp [-R [-H | -L | -P]] [-fi | -n] [-apvXc] source_file ... target_directory
exit status 64
/Users/barrysims/Desktop/Arduino.app/Contents/Java/arduino-builder returned 64
Error compiling for board ESP32 Dev Module.

Any ideas

Bodmer, 2021-01-19:
This looks like an IDE problem rather than a library issue. Have you been able to compile and upload the "Blink" example to an ESP32.
If you select an Arduino UNO Board does it compile?

inb4ohnoes, 2021-01-21:
I'm having this issue as well on esp32 package version 1.0.5 dev builds. To get it to work, I ended up switching the esp32 library version to 1.0.4. However 1.0.4 has the issue with esptool, so I used [this modified esptool binary](https://github.com/Bodmer/TFT_eSPI/files/5847954/esptool.zip) which works around it. This is seemingly an issue with the rc builds of 1.0.5 they've yet to solve.

Bodmer, 2021-01-21:
@inb4ohnoes thanks for your post. I am not familiar with using a MAC so your input is great.

bazfromleek, 2021-01-21:
Hi everyone, thanks for the feedback - sorry for the delay in updating but I solved this. I believe this was because I was using a manual method to install the ESP32 Library with it ending up in the sketch folder. I tried everything from removing Arduino to reinstalling library (again manually) but kept getting this bash error. The reason I was installing manually was because when I used the Library manager to install ESP32 I kept getting a different error re serial and python. Installing manual was the recommendation to fix that. Anyway - long story short, I installed the ESP32 Library through the manager, then sorted the espool.py issue that is well documented, then had to install serial into python. I think that is how I got to a working state. To be honest I spent a lot of hours doing lots of things and not 100% what sorted it lol :-) We have all been there I am sure.

closed 2021-01-21 by bazfromleek