Bodmer/TFT_eSPI #944: Screensize please move definition back to user_setup

arnold-b, 2021-01-19:
ili9488 defines

Hello, 

with various screen size its better to handle if the screensize is configured inside user_setup.h like before

thanks

Bodmer, 2021-01-20:
Sone driver options do permit the defaults to be over-ridden, these are the displays that are often available from vendors with different screen resolutions, e.g. ST7735 and ST7789.

I am not aware that the ILI9488 is available in resolutions other than 320x480. If you have found a vendor then please provide a web link, as this may be of interest to other library users.

In the meantime add the #ifndef lines you propose to[ the file here](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ILI9488_Defines.h#L3-L4).

Bodmer, 2021-01-20:
Forgot to mention that you can use the constructor to modify the screen resolution, e.g.:

TFT_eSPI tft = TFT_eSPI( 320, 320 );

However you will then have an offset issue if you use some screen rotations.  The displays that allow user setup over-rides use offsets to compensate. The ILI9488 software driver does not have any offsets built in, however you can use the setViewport function to make sure the drawing area remains in the visible area.

closed 2021-01-20 by Bodmer

arnold-b, 2021-01-20:
Thanks this i didnt know TFT_eSPI tft = TFT_eSPI( 320, 320 );

paultbarrett, 2021-10-02:
There is now an ili9488 screen which supports 320x320 resolution and having this #define is presenting issues. Can #ifndef now be added as per arnold-b's request. Datasheet attached for reference

[3 92inch 392-B4042-C规格书(带电容TP）.pdf](https://github.com/Bodmer/TFT_eSPI/files/7271364/3.92inch.392-B4042-C.TP.pdf)