Bodmer/TFT_eSPI #946: Support for the RA8875

tonyc770, 2021-01-20:
I'm looking at a 4.3 inch 800x480 TFT with an RA8875 display driver.
Any support for such a device?

Thanks...

Bodmer, 2021-01-20:
No, the RA8875 is a bit of an oddball display as it has built in graphics functions that require special command sequences. It is possible to use it in a pixel by pixel mode but I am told that this results in rather a poor screen update performance.

The SSD1963 is supported and 480x800 displays are available with that controller. I expect you will find the price is similar.

tonyc770, 2021-01-20:
Is there a display with the SSD1963 800X480 that might be sunlight readable that you can recommend?

Thanks

Bodmer, 2021-01-20:
Readability in sunlight requires a very high backlight intensity  (>1000 cd/m2) and these displays are quite rare. You may need to go to a specialist supplier. I do not own any and have not noticed any suitable displays with that resolution  in my past browsing, so I would try Google.

closed 2021-01-20 by Bodmer