Bodmer/TFT_eSPI #947: class TFT_eSPI has no member named 'getTouch'  &&  class TFT_eSPI has no member named 'calibrateTouch'

Mustafa-Montaser, 2021-01-20:
Iam using ESP32 DEVKIT V1 (30 pin) and 2.8 TFT touch screen (parallel interface) with ILI9341 Driver.

Iam trying to run "ON_OFF_Button" example  or **any example containing  touch functionality** and I always get the same error which in following lines :
=======================================================================
src\main.cpp: In function 'void setup()':
src\main.cpp:66:19: error: 'touch_calibrate' was not declared in this scope
   touch_calibrate();
                   ^
src\main.cpp: In function 'void loop()':
src\main.cpp:81:11: error: 'class TFT_eSPI' has no member named 'getTouch'
   if (tft.getTouch(&x, &y))
           ^
src\main.cpp: In function 'void touch_calibrate()':
src\main.cpp:145:9: error: 'class TFT_eSPI' has no member named 'setTouch'
     tft.setTouch(calData);
         ^
src\main.cpp:164:9: error: 'class TFT_eSPI' has no member named 'calibrateTouch'
     tft.calibrateTouch(calData, TFT_MAGENTA, TFT_BLACK, 15);
         ^
*** [.pio\build\esp32doit-devkit-v1\src\main.cpp.o] Error 1
 ===========================================================
and here is the User_Setup.h
=========================
#define ESP32_PARALLEL
#define ILI9341_DRIVER
// ESP32 pins used for the parallel interface TFT
#define TFT_CS   27  // Chip select control pin
#define TFT_DC   14  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  -1 // Reset pin  26
#define TFT_WR   12  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD   13
#define TFT_D0   16  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   4  // so a single register write sets/clears all bits
#define TFT_D2   23
#define TFT_D3   22
#define TFT_D4   21
#define TFT_D5   19
#define TFT_D6   18
#define TFT_D7   17

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT
==================================================================
I don't know a lot about this library . this is first time to use it .
Any help please .

Bodmer, 2021-01-20:
The library does not support touch controllers when a parallel interface is used. Hence the error message.

You can use other support libraries.[ See the discussions here](https://github.com/Bodmer/TFT_eSPI/discussions/categories/q-a-touch).

Bodmer, 2021-01-20:
[See also the library ](https://github.com/s60sc/Adafruit_TouchScreen)here if your display has an analogue resistance touch screen interface.

closed 2021-01-20 by Bodmer

Mustafa-Montaser, 2021-01-22:
> [See also the library ](https://github.com/s60sc/Adafruit_TouchScreen)here if your display has an analogue resistance touch screen interface.

thanks a lot for your reply 
I downloaded that library and it works 
thank you again

Bodmer, 2021-01-22:
Good to hear. Hope your project is a success.