Bodmer/TFT_eSPI #948: No communication via RF24 when getTouch

hm-soft, 2021-01-21:
Hello
Sorry first for my poor english
In my current project I have:
- Wemos D1 mini
- 2,4 " SPI TFT with ILI9341
- a nRF24L01 
Display and touch are working fine, the RF24-Module works fine too as long I do not call function "getTouch" in the loop. When I periodically call getTouch in loop ( I tried also some delays between) nothing more is received via RF24.
This means I can use the touch feature OR the RF24, but not both together.
What can I do? I think it's not a hardware problem, because CE and CSN of RF24 are D2+D0 (or D1) and CSs of TFT and touch are D3+D4.
Any idea?
Thx
Hubi

Bodmer, 2021-01-21:
Which version of TFT_eSPI?

hm-soft, 2021-01-21:
my actual installed version is 2.2.14, not the newest. Maybe is this the problem?

hm-soft, 2021-01-21:
I updated to newest available version 2.3.56 and get the same result, no RF24 receive when polling the touch.

hm-soft, 2021-01-21:
BTW: update removed my user setup header file, not so good!

Bodmer, 2021-01-21:
I will look at the code and see if I can find the problem.  In the meantime[ this may help ](https://github.com/Bodmer/TFT_eSPI#tips)in future.

Bodmer, 2021-01-21:
Which library do you use with the RF24-Module?

Bodmer, 2021-01-21:
Have you enabled transaction support in the setup file? [See here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L331-L336).

If you run the "Read_User_Setup" diagnostic sketch and look at the serial output this will tell you if the settings you expect are being picked up by the compiler.

hm-soft, 2021-01-21:
I'm using the newest version 1.3.11 by TMRh20

hm-soft, 2021-01-21:
your mentioned "Read_used_setup" shows mw following output:
code]
TFT_eSPI ver = 2.3.56
Processor    = ESP8266
Frequency    = 80MHz
Voltage      = 3.29V
Transactions = No
Interface    = SPI
SPI overlap  = No

Display driver = 9341
Display width  = 240
Display height = 320

TFT_CS   = PIN_D8
TFT_DC   = PIN_D3
TOUCH_CS = PIN_D4

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.00
Touch SPI frequency   = 2.50
[/code]

hm-soft, 2021-01-21:
Transaction support is now enabled,but no change, no message via RF24 is arriving

hm-soft, 2021-01-21:
this is my user setup file:

#define ILI9341_DRIVER

#define TFT_CS   PIN_D8  // Chip select control pin D8
#define TFT_DC   PIN_D3  // Data Command control pin
//#define TFT_RST  PIN_D4  // Reset pin (could connect to NodeMCU RST, see next line)
#define TFT_RST  -1    // Set TFT_RST to -1 if the display RESET is connected to NodeMCU RST or 3.3V

#define TOUCH_CS PIN_D4     // Chip select pin (T_CS) of touch screen

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

// #define SPI_FREQUENCY  27000000
#define SPI_FREQUENCY  40000000
// #define SPI_FREQUENCY  80000000

#define SPI_READ_FREQUENCY  20000000

#define SPI_TOUCH_FREQUENCY  2500000

#define SUPPORT_TRANSACTIONS

Bodmer, 2021-01-21:
I think the problem is that [this line must be commented out as shown here](https://github.com/Bodmer/TFT_eSPI/blob/master/Extensions/Touch.cpp#L41).

This has been done in the download from Github, but not in the one currently available with the Arduino IDE download.

Bodmer, 2021-01-22:
I have released version 2.3.59, this fixes the problem. Post back if you still have problems.

closed 2021-01-22 by Bodmer