Bodmer/TFT_eSPI #960: ILI9481 white screen on ESP32

pedro371, 2021-01-27:
Hello,

i have connected a ili9481 on a ESP32 with this settings in the User_Setup.h

`#define ILI9481_DRIVER`

`// Tell the library to use 8 bit parallel mode (otherwise SPI is assumed)
#define TFT_PARALLEL_8_BIT

// The ESP32 and TFT the pins used for testing are:
#define TFT_CS   33  // Chip select control pin (library pulls permanently low
#define TFT_DC   15  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  32  // Reset pin, toggles on startup

#define TFT_WR    4  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD    2  // Read strobe control pin

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits.
#define TFT_D2   26  // Pins can be randomly assigned, this does not affect
#define TFT_D3   25  // TFT screen update performance.
#define TFT_D4   17
#define TFT_D5   16
#define TFT_D6   27
#define TFT_D7   14`

But my screen is always white... do you have an idea what I have done wrong?

[image]
[image]

Bodmer, 2021-01-27:
First run the Read_User_Setup diagnostic sketch, this outputs the configuration values picked up by the compiler and prints output to the Serial Monitor window, nothing is shown on the display. Check the output against your expected settings, post output here if you are not sure.

The simplest way to connect a display to an ESP32 is to use the UNO formay ESP32, see readme. Connecting by wires can be problematic due  to ground bounce. Use short signal wires and at least 2 short GND wires.

pedro371, 2021-01-29:
@Bodmer First of all thanks for your great library!

No it's working but I also have an another problem. How can I use touch in the parallel TFT setup?

Bodmer, 2021-01-29:
You can use a touch support library. The choice is dependant on the type of touch interface your display has.

closed 2021-02-01 by Bodmer